code-iai/iai_kinect2 #230: How to get rgb image projected in IR_Coordinate Frame?

filipetrocadoferreira, 2016-02-22:
How can I get rgb image projected in ir coordinate frame without depth information like is currently implemented. 

Because I was subscribing both qhd topics but I have framerate issues, so I want to have both images rectified in small definition. Which is the best way to get them?

wiedemeyer, 2016-02-22:
You can't project the color image into the ir frame without depth information.

then just subscribe to `/kinect2/sd/image_color_rect` and  `/kinect2/sd/image_depth_rect`.

filipetrocadoferreira, 2016-02-22:
Why not? Sorry, but i really can't understand why

wiedemeyer, 2016-02-22:
I am not sure if I understand your question correctly. But the position of a color pixel in IR frame depends on the depth value. Basically the code for registering a color image to the IR frame takes a depth pixel and computes the position in color space, then it takes the color information from that position and stores it in an image with the same size as the depth image. Maybe you want to look at the implementation https://github.com/OpenKinect/libfreenect2/blob/master/src/registration.cpp#L128.
But if you just want the color image, just subscribe to `/kinect2/sd/image_color_rect`.

filipetrocadoferreira, 2016-02-22:
Ok. I understood that your implementation uses depth information to project (or the right term is register) a color pixel in IR coordinate frame. That's why /kinect2/sd/image_color_rect has some black pixels related to missed depth information

So, now my question is if you think is there any chance to registar rgb camera in IR plane without using depth information. Since we're talking here about a plane to plane transformation that could be solved using homography no? Think about stereo configuration,  usually you project on of the camera to the other's coordinate frame.

filipetrocadoferreira, 2016-02-22:
i guess i'm not making much sense. The task of transforming depth->rgb coodinate frame is totally different of rgb->depth.  So my question is why subscribing /kinect2/qhd/image_depth_rect is quite slower than /kinect2/sd/image_depht_rect . Registration task needs so much computational effort?

wiedemeyer, 2016-02-23:
The registration is quite fast normally. What do you mean with slow? Are you transferring the images over the network? If so it could be bandwidth limitations and you could subscribe to the compressed topics.

filipetrocadoferreira, 2016-02-23:
That was what I wrote here  https://github.com/code-iai/iai_kinect2/issues/196 

I'm running iai-kinect in a i5 without gpu and when I subscribe to /qhd/depth topics the frame rate decreases a lot (to ~6Hz).

wiedemeyer, 2016-02-24:
I guess you main problem is a high CPU usage. The i5 you have has a Intel GPU, you can use it with OpenCL. Try to setup Beignet for OpenCL, as described athttps://github.com/OpenKinect/libfreenect2#linux. There is also a VAAPI implementation for JPEG deconding in libfreenect2 now, this also helps to lower CPU usage. Instructions are also in the README.

filipetrocadoferreira, 2016-02-24:
I'll do that. Libfreenect is very active and I must use latest code. I had problems with opencl and Beignet. I will try the latest code and opencl registration and give you feedback! Thanks

closed 2016-04-14 by wiedemeyer