code-iai/iai_kinect2 #231: could not find a package configuration file provided by "freenect2"

zmartycc, 2016-02-22:
After  the command
**catkin_make -DCMAKE_BUILD_TYPE="Release"**
，there was a error:
**CMake Error at iai_kinect2/kinect2_bridge/CMakeLists.txt:22 (find_package):
  By not providing "Findfreenect2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "freenect2", but CMake did not find one.
  Could not find a package configuration file provided by "freenect2" with
  any of the following names:

    freenect2Config.cmake
    freenect2-config.cmake**.
I don't know how to fix it. Can anyone help me?

wiedemeyer, 2016-02-22:
try to install libfreenect2.

zmartycc, 2016-02-22:
I had installed libfreenect2 as the instructions. I used Kinect for windows v1 and the command ./bin/Protonect didn't work. The output message said no device connected.  Nothing else seemed wrong.

wiedemeyer, 2016-02-22:
And did you do `make install` in libfreenect2? What path did you installed it to?
BTW: It will only work with Kinect v2.

zmartycc, 2016-02-22:
I  did **make install** in **libfreenect2/depends/build**, and **/build** was a new file created followed the instructions.

wiedemeyer, 2016-02-23:
Where did `make install` installed the files to? What is the output on the terminal?

xlz, 2016-02-23:
For iai_kinect2, `catkin_make -DCMAKE_BUILD_TYPE="Release" -Dfreenect2_DIR=$HOME/freenect2`, this is what I have been doing.

wiedemeyer, 2016-02-24:
Fixed it in fa55624ea7074ad66693582f53defb05ab14564b. `{HOME}/freenect2` is now one of the paths where it searches for libfreenect2.

closed 2016-02-29 by wiedemeyer

caozhenn, 2018-12-10:
you should install freenect in your XX_ws instead of home, some tutorials are wrong, try another tutorials.