code-iai/iai_kinect2 #232: CMake error  configuration file provided by "freenect2"

user3e93rf23, 2016-02-22:
Unfortunately I cannot make senso of this error. Where do I get these cmake-files?

```
CMake Error at iai_kinect2/kinect2_bridge/CMakeLists.txt:22 (find_package):
  By not providing "Findfreenect2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "freenect2", but CMake did not find one.

  Could not find a package configuration file provided by "freenect2" with
  any of the following names:

    freenect2Config.cmake
    freenect2-config.cmake

  Add the installation prefix of "freenect2" to CMAKE_PREFIX_PATH or set
  "freenect2_DIR" to a directory containing one of the above files.  If
  "freenect2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/baum/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/baum/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

I did install all related repositoryies like libfreenect2:
```
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/lib/libfreenect2.so.0.2.0
-- Installing: /usr/local/lib/libfreenect2.so.0.2
-- Installing: /usr/local/lib/libfreenect2.so
-- Installing: /usr/local/include/libfreenect2
-- Installing: /usr/local/include/libfreenect2/registration.h
-- Installing: /usr/local/include/libfreenect2/logger.h
-- Installing: /usr/local/include/libfreenect2/packet_pipeline.h
-- Installing: /usr/local/include/libfreenect2/frame_listener_impl.h
-- Installing: /usr/local/include/libfreenect2/frame_listener.hpp
-- Installing: /usr/local/include/libfreenect2/libfreenect2.hpp
-- Installing: /usr/local/include/libfreenect2
-- Installing: /usr/local/include/libfreenect2/export.h
-- Installing: /usr/local/include/libfreenect2/config.h
-- Installing: /usr/local/lib/cmake/freenect2/freenect2Config.cmake
-- Installing: /usr/local/lib/pkgconfig/freenect2.pc
```

wiedemeyer, 2016-02-23:
Hi,
could you build it in debug and start it in gdb to see where is crashes?

user3e93rf23, 2016-02-23:
Yes I followed instrunctions for gdb:

`baum@baum-ThinkPad-L520:~/catkin_ws/devel/lib/kinect2_bridge$ gdb kinect2_bridge
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from kinect2_bridge...done.
(gdb) run
Starting program: /home/baum/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[New Thread 0x7fffdc310700 (LWP 3346)]
[New Thread 0x7fffdac61700 (LWP 3349)]
[New Thread 0x7fffda460700 (LWP 3351)]
[New Thread 0x7fffd9c5f700 (LWP 3352)]
[New Thread 0x7fffd945e700 (LWP 3358)]
[New Thread 0x7fffd3fff700 (LWP 3359)]
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/baum/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4

Program received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/x86_64/strlen.S:106
106	../sysdeps/x86_64/strlen.S: No such file or directory.
(gdb) bt
#0  strlen () at ../sysdeps/x86_64/strlen.S:106
#1  0x00007ffff4b6d82e in __GI___strdup (s=0x0) at strdup.c:41
#2  0x00007fffe693d816 in ?? () from /lib/x86_64-linux-gnu/libudev.so.1
#3  0x00007fffe693f359 in ?? () from /lib/x86_64-linux-gnu/libudev.so.1
#4  0x00007fffe693f5f4 in udev_device_new_from_syspath () from /lib/x86_64-linux-gnu/libudev.so.1
#5  0x00007fffd826c707 in udev_device_new_from_devnum () from /lib/x86_64-linux-gnu/libudev.so.0
#6  0x00007fffee654be4 in ?? () from /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1
#7  0x00007fffee65507a in ?? () from /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1
#8  0x00007fffee64e4fc in ?? () from /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1
#9  0x00007fffee62bd9c in ?? () from /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1
#10 0x00007fffee6279e1 in glXQueryVersion () from /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1
#11 0x00007ffff5713d34 in _glfwInitGLX () from /usr/local/lib/libfreenect2.so.0.2
#12 0x00007ffff570c451 in _glfwPlatformInit () from /usr/local/lib/libfreenect2.so.0.2
#13 0x00007ffff5704c34 in glfwInit () from /usr/local/lib/libfreenect2.so.0.2
#14 0x00007ffff56e81a5 in Timer (this=0x6c1640)
    at /home/baum/catkin_ws/src_3party/libfreenect2/src/logging.cpp:203
#15 WithPerfLoggingImpl (this=0x6c1640)
    at /home/baum/catkin_ws/src_3party/libfreenect2/src/logging.cpp:257
#16 libfreenect2::WithPerfLogging::WithPerfLogging (this=0x70b340)
    at /home/baum/catkin_ws/src_3party/libfreenect2/src/logging.cpp:313
#17 0x00007ffff570289a in libfreenect2::OpenCLDepthPacketProcessorImpl::OpenCLDepthPacketProcessorImpl (this=0x70b340, deviceId=-1)
    at /home/baum/catkin_ws/src_3party/libfreenect2/src/opencl_depth_packet_processor.cpp:236
#18 0x00007ffff56fc9e1 in libfreenect2::OpenCLDepthPacketProcessor::OpenCLDepthPacketProcessor (
    this=0x709e60, deviceId=-1)
    at /home/baum/catkin_ws/src_3party/libfreenect2/src/opencl_depth_packet_processor.cpp:717
#19 0x00007ffff56e0e26 in libfreenect2::OpenCLPacketPipeline::OpenCLPacketPipeline (this=0x6c1410, 
    deviceId=-1) at /home/baum/catkin_ws/src_3party/libfreenect2/src/packet_pipeline.cpp:151
#20 0x000000000044f996 in Kinect2Bridge::initPipeline (this=0x7fffffffca70, method=..., device=-1)
    at /home/baum/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:403
#21 0x000000000044e535 in Kinect2Bridge::initialize (this=0x7fffffffca70)
    at /home/baum/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:294
#22 0x000000000044c125 in Kinect2Bridge::start (this=0x7fffffffca70)
    at /home/baum/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:146
#23 0x00000000004478f0 in main (argc=1, argv=0x7fffffffd9d8)
    at /home/baum/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1490
(gdb) quit
`

user3e93rf23, 2016-02-23:
I also get this during debug build:
`baum@baum-ThinkPad-L520:~/catkin_ws$            from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
from: can't read /var/mail//usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
baum@baum-ThinkPad-L520:~/catkin_ws$                  from /home/baum/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:32:
from: can't read /var/mail//home/baum/catkin_ws/src/iai_kinect2/kinect2_viewer/src/viewer.cpp:32:
baum@baum-ThinkPad-L520:~/catkin_ws$ /usr/include/pcl-1.7/pcl/visualization/area_picking_event.h: In member function ‘bool pcl::visualization::AreaPickingEvent::getPointsIndices(std::vector<int>&) const’:
bash: syntax error near unexpected token `('
baum@baum-ThinkPad-L520:~/catkin_ws$ /usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
bash: /usr/include/pcl-1.7/pcl/visualization/area_picking_event.h:64:30:: No such file or directory
baum@baum-ThinkPad-L520:~/catkin_ws$            if (nb_points_ == -1)
>                               ^
`

user3e93rf23, 2016-02-23:
to be sure i reinstalled libfreenect2 in this configuration:

[image]

and make install does outputs:
`Install the project...
-- Install configuration: "RelWithDebInfo"
-- Up-to-date: /home/baum/freenect2/lib/libfreenect2.so.0.2.0
-- Up-to-date: /home/baum/freenect2/lib/libfreenect2.so.0.2
-- Up-to-date: /home/baum/freenect2/lib/libfreenect2.so
-- Installing: /home/baum/freenect2/include/libfreenect2
-- Up-to-date: /home/baum/freenect2/include/libfreenect2/registration.h
-- Up-to-date: /home/baum/freenect2/include/libfreenect2/logger.h
-- Up-to-date: /home/baum/freenect2/include/libfreenect2/packet_pipeline.h
-- Up-to-date: /home/baum/freenect2/include/libfreenect2/frame_listener_impl.h
-- Up-to-date: /home/baum/freenect2/include/libfreenect2/frame_listener.hpp
-- Up-to-date: /home/baum/freenect2/include/libfreenect2/libfreenect2.hpp
-- Installing: /home/baum/freenect2/include/libfreenect2
-- Up-to-date: /home/baum/freenect2/include/libfreenect2/export.h
-- Up-to-date: /home/baum/freenect2/include/libfreenect2/config.h
-- Up-to-date: /home/baum/freenect2/lib/cmake/freenect2/freenect2Config.cmake
-- Up-to-date: /home/baum/freenect2/lib/pkgconfig/freenect2.pc
`

user3e93rf23, 2016-02-23:
Protonect is not building, I will suspend this and report later

user3e93rf23, 2016-02-23:
libfreenect2 got build and is ok via proton gl check. <br> I am stuck under install point4: there is no rules inside freenect2 ( only include and lib). So it says no such file or directory

user3e93rf23, 2016-02-23:
I cannot make sense of this and dont know which direction to go:

> CMake Error at iai_kinect2/kinect2_bridge/CMakeLists.txt:22 (find_package):
  By not providing "Findfreenect2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "freenect2", but CMake did not find one.

  Could not find a package configuration file provided by "freenect2" with
  any of the following names:

    freenect2Config.cmake
    freenect2-config.cmake

  Add the installation prefix of "freenect2" to CMAKE_PREFIX_PATH or set
  "freenect2_DIR" to a directory containing one of the above files.  If
  "freenect2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!

user3e93rf23, 2016-02-23:
mmh. I do have the first one from above under` freenect2/lib/cmake/freenect2`

user3e93rf23, 2016-02-23:
I used ccmake . to solve it. Now I cannot generate because of cmake warning with eigen

>  CMake Warning at /opt/ros/jade/share/cmake_modules/cmake/Modules/FindEigen.cmake:62 (message):
   The FindEigen.cmake Module in the cmake_modules package is deprecated.

   Please use the FindEigen3.cmake Module provided with Eigen.  Change
   instances of find_package(Eigen) to find_package(Eigen3).  Check the
   FindEigen3.cmake Module for the resulting CMake variable names.

 Call Stack (most recent call first):
   iai_kinect2/kinect2_registration/CMakeLists.txt:31 (find_package)

user3e93rf23, 2016-02-23:
I think I got it also I almost lost it, Wow. I have no Idea why the installation directory of libfreenect2 is not simply a module_path where cmake is looking by default, so I had to copy it into /lib eg. where cmake looks for [default](https://cmake.org/cmake/help/v2.8.8/cmake.html#command%3afind_package).

user3e93rf23, 2016-02-23:
I am still wondering what point4 is for:

> I am stuck under install point4: there is no rules inside freenect2 ( only include and lib). 
Is it ok to have skipped it as I dont have the directory

user3e93rf23, 2016-02-23:
> 
> `baum@baum-ThinkPad-L520:~/catkin_ws$ roslaunch kinect2_bridge kinect2_bridge.launch
> ... logging to /home/baum/.ros/log/71303404-da46-11e5-b8b6-74e50b7d17bc/roslaunch-baum-ThinkPad-L520-12298.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://baum-ThinkPad-L520:36684/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /kinect2_bridge/base_name: kinect2
>  * /kinect2_bridge/base_name_tf: kinect2
>  * /kinect2_bridge/bilateral_filter: True
>  * /kinect2_bridge/calib_path: /home/baum/catkin...
>  * /kinect2_bridge/depth_device: -1
>  * /kinect2_bridge/depth_method: default
>  * /kinect2_bridge/edge_aware_filter: True
>  * /kinect2_bridge/fps_limit: -1.0
>  * /kinect2_bridge/jpeg_quality: 90
>  * /kinect2_bridge/max_depth: 12.0
>  * /kinect2_bridge/min_depth: 0.1
>  * /kinect2_bridge/png_level: 1
>  * /kinect2_bridge/publish_tf: False
>  * /kinect2_bridge/queue_size: 5
>  * /kinect2_bridge/reg_devive: -1
>  * /kinect2_bridge/reg_method: default
>  * /kinect2_bridge/sensor: 
>  * /kinect2_bridge/use_png: False
>  * /kinect2_bridge/worker_threads: 4
>  * /kinect2_points_xyzrgb_hd/queue_size: 5
>  * /kinect2_points_xyzrgb_qhd/queue_size: 5
>  * /kinect2_points_xyzrgb_sd/queue_size: 5
>  * /rosdistro: jade
>  * /rosversion: 1.11.16
> 
> NODES
>   /
>     kinect2 (nodelet/nodelet)
>     kinect2_bridge (nodelet/nodelet)
>     kinect2_points_xyzrgb_hd (nodelet/nodelet)
>     kinect2_points_xyzrgb_qhd (nodelet/nodelet)
>     kinect2_points_xyzrgb_sd (nodelet/nodelet)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> core service [/rosout] found
> process[kinect2-1]: started with pid [12316]
> process[kinect2_bridge-2]: started with pid [12317]
> process[kinect2_points_xyzrgb_sd-3]: started with pid [12318]
> process[kinect2_points_xyzrgb_qhd-4]: started with pid [12319]
> process[kinect2_points_xyzrgb_hd-5]: started with pid [12326]
> [ INFO] [1456250031.852324821]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
> [ INFO] [1456250031.861009240]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
> [ INFO] [1456250031.898318526]: Initializing nodelet with 4 worker threads.
> [ INFO] [1456250031.906303124]: waitForService: Service [/kinect2/load_nodelet] is now available.
> [ INFO] [1456250032.348491956]: [Kinect2Bridge::initialize] parameter:
>         base_name: kinect2
>            sensor: default
>         fps_limit: -1
>        calib_path: /home/baum/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
>           use_png: false
>      jpeg_quality: 90
>         png_level: 1
>      depth_method: default
>      depth_device: -1
>        reg_method: default
>        reg_devive: -1
>         max_depth: 12
>         min_depth: 0.1
>        queue_size: 5
>  bilateral_filter: true
> edge_aware_filter: true
>        publish_tf: false
>      base_name_tf: kinect2
>    worker_threads: 4
> [Info] [Freenect2Impl] enumerating devices...
> [Info] [Freenect2Impl] 8 usb devices connected
> [Info] [Freenect2Impl] found 0 devices
> [ERROR] [1456250032.540770761]: [Kinect2Bridge::initDevice] no Kinect2 devices found!
> [ERROR] [1456250032.543927698]: [Kinect2Bridge::start] Initialization failed!
> [FATAL] [1456250032.599816322]: Service call failed!
> [kinect2_bridge-2] process has died [pid 12317, exit code 255, cmd /opt/ros/jade/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/baum/.ros/log/71303404-da46-11e5-b8b6-74e50b7d17bc/kinect2_bridge-2.log].
> log file: /home/baum/.ros/log/71303404-da46-11e5-b8b6-74e50b7d17bc/kinect2_bridge-2*.log
> [kinect2_bridge-2] restarting process
> process[kinect2_bridge-2]: started with pid [12499]
> [ INFO] [1456250032.930016905]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
> [ INFO] [1456250032.976143775]: [Kinect2Bridge::initialize] parameter:
>         base_name: kinect2
>            sensor: default
>         fps_limit: -1
>        calib_path: /home/baum/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
>           use_png: false
>      jpeg_quality: 90
>         png_level: 1
>      depth_method: default
>      depth_device: -1
>        reg_method: default
>        reg_devive: -1
>         max_depth: 12
>         min_depth: 0.1
>        queue_size: 5
>  bilateral_filter: true
> edge_aware_filter: true
>        publish_tf: false
>      base_name_tf: kinect2
>    worker_threads: 4
> [Info] [Freenect2Impl] enumerating devices...
> [Info] [Freenect2Impl] 8 usb devices connected
> [Info] [Freenect2Impl] found 0 devices
> [ERROR] [1456250033.132945350]: [Kinect2Bridge::initDevice] no Kinect2 devices found!
> [ERROR] [1456250033.135766289]: [Kinect2Bridge::start] Initialization failed!
> [FATAL] [1456250033.139175826]: Service call failed!
> [kinect2_bridge-2] process has died [pid 12499, exit code 255, cmd /opt/ros/jade/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/baum/.ros/log/71303404-da46-11e5-b8b6-74e50b7d17bc/kinect2_bridge-2.log].
> log file: /home/baum/.ros/log/71303404-da46-11e5-b8b6-74e50b7d17bc/kinect2_bridge-2*.log
> [kinect2_bridge-2] restarting process
> `

wiedemeyer, 2016-02-23:
Of course, you have to install it somewhere where cmake looks for stuff, for example `/usr/local/`. I just saw, they changed the install instructions, `${HOME}/freenect2` will not work, previously it defaulted to `/usr/local/`
So finally it seems to work. Now you only need to connect the Kinect2 to a USB 3 port. Also make sure to that you have done: "Set up udev rules for device access: `sudo cp ../platform/linux/udev/90-kinect2.rules /etc/udev/rules.d/`, then replug the Kinect."

jsr38, 2016-02-23:
Oh what a shitstorm you create.

user3e93rf23, 2016-02-23:
As I said there how to copy if the directory is not existent. Where do you want me to look for the udev rules?

JimmyDaSilva, 2016-02-23:
Not that hard to find seriously...
https://github.com/OpenKinect/libfreenect2/blob/master/platform/linux/udev/90-kinect2.rules

wiedemeyer, 2016-02-23:
https://github.com/OpenKinect/libfreenect2/blob/master/platform/linux/udev/90-kinect2.rules

user3e93rf23, 2016-02-24:
Oh, I was looking inside the freenect2 directory. Sorry<br> Thank you for your help it does work fine now. In case anyone wants to know ahead of time which topics are published right now I added a screenshot:
[image]

closed 2016-02-24 by user3e93rf23