code-iai/iai_kinect2 #233: cannot save the frame

MF1523017, 2016-02-24:
After this command
**rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color,**
Everything is ok,but using space or s  cannot save the frame to images ,the Esc or q can quit.
Can anyone help me?

 [ INFO] [main] Start settings:
       Mode: record
     Source: color
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] starting recorder...
[ INFO] [Recorder::startRecord] Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decrease min and max value for IR value rage
        [h] - increase min and max value for IR value rage
        [1] - decrease min value for IR value rage
        [2] - increase min value for IR value rage
        [3] - decrease max value for IR value rage
        [4] - increase max value for IR value rage
[ INFO] [main] stopped recording...

JimmyDaSilva, 2016-02-24:
@MF1523017 You can try to click on the terminal window where you launch this node before pressing space. Or click on the image window then press. I don't remember on which window you should be before pressing the keys

MF1523017, 2016-02-24:
@JimmyDaSilva I have solved the problem, only when the image window contains chessboard can I save the frame. Thank you all the same!

JimmyDaSilva, 2016-02-24:
@MF1523017 Sure no problem. Please close the issue then.

closed 2016-02-24 by MF1523017