code-iai/iai_kinect2 #236: OpenCL Error 57 on Depth Registration

guanchar, 2016-02-26:
This error occurs while running kinect_bridge with openCL depth registration (with connected clients). Kinect_viewer shows no image/point cloud.

`[ERROR] [1456462003.973289943]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](293) data->queue.enqueueReadBuffer(data->bufferRegistered, CL_FALSE, 0, data->sizeRegistered, data->dataOutput, &eventCheckDepth2, &eventRead) failed: -57`

This error continuously outputs, while occasionally the below is printed:

`[ INFO] [1456462003.670611197]: [Kinect2Bridge::main] depth processing: ~5.67733ms (~176.139Hz) publishing rate: ~30.3021Hz
[ INFO] [1456462003.670666884]: [Kinect2Bridge::main] color processing: ~13.4652ms (~74.2653Hz) publishing rate: ~14.9845Hz`

The system is Ubuntu 14.04, with a 4.4.1 kernel, although 4.3.3 doesn't seem to work either. All steps were carried out on fresh system install.

Protonect runs fine on with openCL processing. The instructions for installing libfreenect were followed as of Feb 25, 2016.

Error 57 is CL_INVALID_ EVENT_WAIT_LIST. 

Beignet and openCL installations:
`ii  beignet-dev                            1.0.3-0pmjdebruijn1~trusty   amd64  `
`ii  beignet-opencl-icd:amd64     1.0.3-0pmjdebruijn1~trusty   amd64 `
`ii  ocl-icd-libopencl1:amd64           2.2.8-1             amd64  `
`ii  ocl-icd-opencl-dev:amd64             2.2.8-1                   amd64 `

Any help in solving this would be greatly appreciated, thank you to @xlz for help so far!

xlz, 2016-02-26:
@wiedemeyer may want to weigh in.

wiedemeyer, 2016-02-26:
I don't see, why it should return `CL_INVALID_ EVENT_WAIT_LIST`. The only reason can be that the event in `eventCheckDepth2` is not valid, but that would mean that the `enqueueNDRangeKernel` before should have failed.
What is the whole output until that error?

guanchar, 2016-02-26:
```
roslaunch kinect2_bridge kinect2_bridge.launch
... logging to /home/algol/.ros/log/78340246-dca1-11e5-950f-fcaa14dbce08/roslaunch-algol-brix4770r-2387.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://algol-brix4770r:40462/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/algol/catki...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_devive: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [2399]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 78340246-dca1-11e5-950f-fcaa14dbce08
process[rosout-1]: started with pid [2412]
started core service [/rosout]
process[kinect2-2]: started with pid [2415]
process[kinect2_bridge-3]: started with pid [2416]
process[kinect2_points_xyzrgb_sd-4]: started with pid [2417]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [2418]
process[kinect2_points_xyzrgb_hd-6]: started with pid [2419]
[ INFO] [1456502177.234341958]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1456502177.238403708]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1456502177.246907303]: Initializing nodelet with 8 worker threads.
[ INFO] [1456502177.259483454]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1456502177.461469257]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/algol/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
[... 39 lines cut ...]
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ INFO] [1456502179.457515056]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1456502179.457623784]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1456502179.466219915]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1456502179.466261502]: [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics Haswell CRW GT3 Desktop (GPU)[Intel]
[ INFO] [1456502179.466288224]: [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics Haswell CRW GT3 Desktop (GPU)[Intel]
[ INFO] [1456502179.552101049]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1456502179.552156158]: [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics Haswell CRW GT3 Desktop (GPU)[Intel]
[ INFO] [1456502179.552175240]: [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics Haswell CRW GT3 Desktop (GPU)[Intel]
[ INFO] [1456502179.632685180]: [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [1456502196.279297438]: [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ERROR] [1456502196.641225293]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](293) data->queue.enqueueReadBuffer(data->bufferRegistered, CL_FALSE, 0, data->sizeRegistered, data->dataOutput, &eventCheckDepth2, &eventRead) failed: -57
[ERROR] [1456502196.688922194]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](293) data->queue.enqueueReadBuffer(data->bufferRegistered, CL_FALSE, 0, data->sizeRegistered, data->dataOutput, &eventCheckDepth2, &eventRead) failed: -57
[ERROR] [1456502196.725378194]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](293) data->queue.enqueueReadBuffer(data->bufferRegistered, CL_FALSE, 0, data->sizeRegistered, data->dataOutput, &eventCheckDepth2, &eventRead) failed: -57
[ERROR] [1456502196.758019454]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](293) data->queue.enqueueReadBuffer(data->bufferRegistered, CL_FALSE, 0, data->sizeRegistered, data->dataOutput, &eventCheckDepth2, &eventRead) failed: -57
[ERROR] [1456502196.788473375]: [DepthRegistrationOpenCL::registerDepth] [depth_registration_opencl.cpp](293) data->queue.enqueueReadBuffer(data->bufferRegistered, CL_FALSE, 0, data->sizeRegistered, data->dataOutput, &eventCheckDepth2, &eventRead) failed: -57
```

hyderjake, 2016-02-28:
I started to have the same error, the kinect_viewer sd works but qhd and hd have blank windows

wiedemeyer, 2016-02-29:
Do you also use Beignet?

xlz, 2016-02-29:
Is it because previously it was not checked?

wiedemeyer, 2016-02-29:
No, the OpenCL calls were also checked before. What I changed was to use a pinned output buffer and I removed the `event.wait()` and instead pass the events to wait for directly to the OpenCL calls. But it is very similar to the libfreenect2 OpenCL implementation, and it works fine for me. I think I will install Beignet and test if it gives the same error for me.
@guanchar @hyderjake could you try out which older version work for you. Just replace depth_registration_opencl.cpp with one of these:
Before pinned memory: https://github.com/code-iai/iai_kinect2/blob/ecec9a65932204017a8d7bf34d26763b0894d7cc/kinect2_registration/src/depth_registration_opencl.cpp
Before event changes: https://github.com/code-iai/iai_kinect2/blob/f2151a6b6c05569ab56547ad05aa5c071715eeeb/kinect2_registration/src/depth_registration_opencl.cpp

wiedemeyer, 2016-02-29:
@xlz 
I installed beignet and can confirm the issue. But now comes the part that I don't understand, I changed
```
CHECK_CL_RETURN(data->queue.enqueueReadBuffer(data->bufferRegistered, CL_FALSE, 0, data->sizeRegistered, data->dataOutput, &eventCheckDepth2, &eventRead));
```
to
```
CHECK_CL_RETURN(eventCheckDepth2[0].wait());
CHECK_CL_RETURN(data->queue.enqueueReadBuffer(data->bufferRegistered, CL_FALSE, 0, data->sizeRegistered, data->dataOutput, NULL, &eventRead));
```
and it works without problems. Even more strage is the fact that if the program is compiled in debug, it works fine with both versions, it only gives an error when compiled in release or release with debug info.

I also noticed that the profiling information for writing the input to the GPU is totally wrong, it reports something like `1.8e+13` ms. (same for the cl profiling in libfreenect2).

wiedemeyer, 2016-02-29:
I pushed a workaround for it to the devel branch.

xlz, 2016-02-29:
I built beignet 1.1.1 and it worked.

wiedemeyer, 2016-02-29:
with the version before the fix, right? Then it was probably a bug in Beignet that has been fixed. So I can remove the workaround later. Thanks for testing. But it is funny, that there were no problems with libfreenect2 were I call it the same way.

xlz, 2016-02-29:
I am unable to pinpoint the root cause. Beignet's build system breaks git bisection.

Anyway, we need a build beignet ppa for Ubuntu 14.04.

wiedemeyer, 2016-03-15:
Beignet 1.1.1 is now available at `ppa:floe/beignet`.

closed 2016-03-15 by wiedemeyer