code-iai/iai_kinect2 #237: Depth calibration failed with OpenCV 3

Butakus, 2016-02-29:
I'm getting an OpenCV error when executing the depth calibration with `rosrun kinect2_calibration kinect2_calibration chess9x6x0.024 calibrate depth`.

I'm using OpenCV 3.1. The other calibration functions work fine. I also tested everything with OpenCV 2.4 with the same files and I'm able to complete the calibration. This error only appears with OpenCV 3.

```
[ INFO] [DepthCalibration::readFiles] restoring file: 0098
[ INFO] [main] starting calibration...
[ INFO] [DepthCalibration::calibrate] frame: ./0000_depth.png
OpenCV Error: Assertion failed (confidence > 0 && confidence < 1) in run, file /home/me/Downloads/opencv-3.1.0/modules/calib3d/src/ptsetreg.cpp, line 178
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/me/Downloads/opencv-3.1.0/modules/calib3d/src/ptsetreg.cpp:178: error: (-215) confidence > 0 && confidence < 1 in function run

Aborted (core dumped)
```

I tracked the error to the file _kinect2_calibration.cpp_, lines 1054-1058:

```cpp
#if CV_MAJOR_VERSION == 2
    cv::solvePnPRansac(board, points[index], cameraMatrix, distortion, rvec, translation, false, 300, 0.05, board.size(), cv::noArray(), cv::ITERATIVE);
#elif CV_MAJOR_VERSION == 3
    cv::solvePnPRansac(board, points[index], cameraMatrix, distortion, rvec, translation, false, 300, 0.05, board.size(), cv::noArray(), cv::SOLVEPNP_ITERATIVE);
#endif
```
And I realized the the function `cv::solvePnPRansac` has different parameters in [OpenCV 2](http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#solvepnpransac) and [OpenCV 3](http://docs.opencv.org/3.0-beta/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#solvepnpransac).
More concretely, the 10th parameter is different. In OpenCV 2.4 it is `int minInliersCount=100`, which is a "big" number. However, in OpenCV 3 it is `double confidence=0.99`, hence the assertion that checks if the confidence is between 0 and 1 fails.

I was able to solve the issue by modifying line 1057 of _kinect2_calibration.cpp_:
```cpp
    cv::solvePnPRansac(board, points[index], cameraMatrix, distortion, rvec, translation, false, 300, 0.05, board.size(), cv::noArray(), cv::SOLVEPNP_ITERATIVE);
```
to
```cpp
    cv::solvePnPRansac(board, points[index], cameraMatrix, distortion, rvec, translation, false, 300, 0.05, 0.99, cv::noArray(), cv::SOLVEPNP_ITERATIVE);
```

I'm passing 0.99 as confidence, instead of board.size(). However, I still don't know which confidence value should be chosen.

Is anyone else getting this error? I would appreciate if someone can replicate it.

wiedemeyer, 2016-02-29:
Hi,
thanks for the issue. I will fix the `cv::solvePnPRansac` call. But OpenCV 3 is not supported at the moment, because there are issues when using OpenCV 2 and 3 together. Is the kinect2_bridge running fine for you with OpenCV 3?

Butakus, 2016-02-29:
I got errors (related to nodelet, I think) using the launch file provided, but I can execute it directly with `rosrun`:
```
rosrun kinect2_bridge kinect2_bridge _depth_method:=opengl _reg_method:=cpu
```
and it works fine.

I know OpenCV 3 is a nightmare, and I understand why it is not supported (I didn't know before). Thank you anyway.
For me, you can close the issue if there is no more to add.

wiedemeyer, 2016-02-29:
fixed with 8315130933c60389956821baa091be6791e4a92c.

closed 2016-02-29 by wiedemeyer