code-iai/iai_kinect2 #240: Problem with kinect2_bridge and libfreenect

AlexxxBeck, 2016-03-04:
Hello,

I am quite new to ROS and want to work with the Kinect 2. I followed your instructions and installed the libfreenect2. That worked fine for all steps. I am also able to run Protonect as well as the optional OpenNI2 as stated on the instructions for libfreenect2.

For the IAI Kinect 2 i followed your instructions by first enabling C++11 and copying the udev rule file. Then i cloned the link and executed rosdep what worked well too.
But when I enter the: catkin_make -DCMAKE_BUILD_TYPE="Release" I get the following error:

faps@FAPSROS:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/faps/catkin_ws
Source space: /home/faps/catkin_ws/src
Build space: /home/faps/catkin_ws/build
Devel space: /home/faps/catkin_ws/devel
Install space: /home/faps/catkin_ws/install
####
#### Running command: "cmake /home/faps/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/faps/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/faps/catkin_ws/install -G Unix Makefiles" in "/home/faps/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/faps/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/faps/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/faps/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/faps/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - beginner_tutorials
-- ~~  - foobar
-- ~~  - ros101
-- ~~  - kinect2_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - kinect2_viewer
-- ~~  - agitr
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'beginner_tutorials'
-- ==> add_subdirectory(beginner_tutorials)
-- Using these message generators: gencpp;genlisp;genpy
-- beginner_tutorials: 1 messages, 1 services
-- +++ processing catkin package: 'foobar'
-- ==> add_subdirectory(foobar)
-- +++ processing catkin package: 'ros101'
-- ==> add_subdirectory(ros101)
-- +++ processing catkin package: 'kinect2_registration'
-- ==> add_subdirectory(iai_kinect2/kinect2_registration)
-- Eigen found (include: /usr/include/eigen3)
-- CPU based depth registration enabled
-- OpenCL based depth registration enabled
-- +++ processing catkin package: 'kinect2_bridge'
-- ==> add_subdirectory(iai_kinect2/kinect2_bridge)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'kinect2_calibration'
-- ==> add_subdirectory(iai_kinect2/kinect2_calibration)
-- +++ processing catkin package: 'kinect2_viewer'
-- ==> add_subdirectory(iai_kinect2/kinect2_viewer)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
[... 677 lines cut ...]
                 from /home/faps/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:50:
/usr/local/include/libfreenect2/frame_listener.hpp:58:3: note: libfreenect2::Frame::Frame(size_t, size_t, size_t)
   Frame(size_t width, size_t height, size_t bytes_per_pixel) :
   ^
/usr/local/include/libfreenect2/frame_listener.hpp:58:3: note:   candidate expects 3 arguments, 4 provided
/usr/local/include/libfreenect2/frame_listener.hpp:40:24: note: constexpr libfreenect2::Frame::Frame(const libfreenect2::Frame&)
 class LIBFREENECT2_API Frame
                        ^
/usr/local/include/libfreenect2/frame_listener.hpp:40:24: note:   candidate expects 1 argument, 4 provided
/home/faps/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1072:18: error: ‘class libfreenect2::Frame’ has no member named ‘format’
       colorFrame.format = libfreenect2::Frame::BGRX;
                  ^
/home/faps/catkin_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1072:27: error: ‘BGRX’ is not a member of ‘libfreenect2::Frame’
       colorFrame.format = libfreenect2::Frame::BGRX;
                           ^
make[2]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o] Fehler 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/all] Fehler 2
make[1]: *** Auf noch nicht beendete Prozesse wird gewartet …
make[2]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o] Fehler 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Fehler 2
make: *** [all] Fehler 2
Invoking "make -j4 -l4" failed
faps@FAPSROS:~/catkin_ws$ 

I'd be really thankful if someone might help me since i tried for the last days to solve the issue but unfortunately couldn't.

wiedemeyer, 2016-03-04:
Hi,
OpenCV 2.4.9 should be fine. The main problem is that something with libfreenect2 is wrong. Which version of libfreenect2 do you have installed? Maybe it is really old or from some fork?

AlexxxBeck, 2016-03-07:
Hi,
i actually did install the newest version of the libfreenect2. Since I had no idea what caused this issue I reinstalled ROS then followed all of your steps and everything worked perfectly. Thank you very much for your fast response though.

closed 2016-03-07 by wiedemeyer