code-iai/iai_kinect2 #241: Question: Depth_colored.png

hyderjake, 2016-03-04:
I'm using your devel branch, and looking in Kinect_viewer, I was wondering if there was a numerical association with the colors in the depth color image and the actual depth(with respect to the calibration)?

snowlyDay, 2016-03-06:
sorry i'm just a new gay for this, i just know when you closer to kinect and the color more deep when you far while it come to white.
And in actually kinect can measure the true depth. there  is X Y Z  in your code can get from kinect.so you can get the  depth.

wiedemeyer, 2016-03-07:
@hyderjake 
The colors are just a visualization of the depth measurements. There is no "depth color image", there is just a depth image with distances in mm for each pixel. The colors are being generated with http://docs.opencv.org/2.4.8/modules/contrib/doc/facerec/colormaps.html.
@snowlyDay 
The depth image contains the depth measurements, together with the camera intrinsic parameters you can compute xyz, or you could just subscribe to the point cloud topic directly.

hyderjake, 2016-03-07:
That is helpful, but just to clarify, if I converted a pixel to a value it would give me the depth in mm?

wiedemeyer, 2016-03-07:
the `*_depth.png` is a 16bit image, each pixel consists of one value with is an `unsigned short`. There shouldn't be any color. The pixels value is the depth in mm. You can use OpenCV to read it in.

hyderjake, 2016-03-07:
The color value was just a gray scale which is the default for opencv, I was trying to do this as simply as possible, I used imagemagick to convert to a txt file and read it directly. I was curious if that file was converted correctly and if that number seemed accurate? I will try it with openCV though

closed 2016-03-29 by hyderjake