code-iai/iai_kinect2 #242: Question- kinect2_bridge with intel Skylake processor

cnanda-zbr, 2016-03-04:
Hi,
Has anyone tested the kinect2_bridge with the intel i7 6th generation processor yet ? We are considering( forced to )  upgrading our platform and want to know how much work it would take to port the code over to this processor. 
We are considering Intel i7-6700TE to be precise, but non TE version should behave the same way I think. 
In general, what is the compatibility of Beignet, OpenCL etc, if some one has tested that already .

-Best
cn

wiedemeyer, 2016-03-07:
There is nothing that needs to be ported, just compile it and you are fine. I don't see any reasons for problems when running it on skylake.
I don't know if Beignet supports skylakes GPU already, you might want to check their website/github/etc.

wiedemeyer, 2016-04-14:
right now there seems to be some (probably fixable) problems with beignet 1.1.1: #248

closed 2016-04-14 by wiedemeyer