code-iai/iai_kinect2 #243: Error in calibrating the depth measurements

AlexxxBeck, 2016-03-07:
Hello,
I unfortunately stuck at calibrating the Kinect. Steps 1-8 worked perfectly fine for me. I recorded and calibrated 150 color, 150 IR and 150 sync images. For Step 9 I wanted to calibrate the depth and it shows me the error that I have no distance sample data. 

At my first try I just recorded and calibrated 90 pictures each for the color, IR and sync images and I stuck on the same problem that I had no distance sample data. So I checked the Issues and someone else already had this Issue before. He said that he just recorded more pictures and then it worked. But after I recorded 150 pictures more for each color, IR and sync images I still get the error that there is no distance sample data.

This is the output of the terminal:
[ INFO] [DepthCalibration::readFiles] restoring file: 0049_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0087_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0125_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0012_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0050
[ INFO] [DepthCalibration::readFiles] restoring file: 0088
[ INFO] [DepthCalibration::readFiles] restoring file: 0126
[ INFO] [DepthCalibration::readFiles] restoring file: 0013
[ INFO] [DepthCalibration::readFiles] restoring file: 0050_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0088_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0126_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0013_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0051
[ INFO] [DepthCalibration::readFiles] restoring file: 0089
[ INFO] [DepthCalibration::readFiles] restoring file: 0127
[ INFO] [DepthCalibration::readFiles] restoring file: 0014
[ INFO] [DepthCalibration::readFiles] restoring file: 0051_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0089_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0127_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0014_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0052
[ INFO] [DepthCalibration::readFiles] restoring file: 0090
[ INFO] [DepthCalibration::readFiles] restoring file: 0128
[ INFO] [DepthCalibration::readFiles] restoring file: 0015
[ INFO] [DepthCalibration::readFiles] restoring file: 0052_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0090_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0128_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0015_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0053
[ INFO] [DepthCalibration::readFiles] restoring file: 0091
[ INFO] [DepthCalibration::readFiles] restoring file: 0129
[ INFO] [DepthCalibration::readFiles] restoring file: 0016
[ INFO] [DepthCalibration::readFiles] restoring file: 0053_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0091_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0129_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0016_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0054
[ INFO] [DepthCalibration::readFiles] restoring file: 0092
[ INFO] [DepthCalibration::readFiles] restoring file: 0130
[ INFO] [DepthCalibration::readFiles] restoring file: 0017
[ INFO] [DepthCalibration::readFiles] restoring file: 0054_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0092_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0130_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0017_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0055
[ INFO] [DepthCalibration::readFiles] restoring file: 0093
[ INFO] [DepthCalibration::readFiles] restoring file: 0131
[ INFO] [DepthCalibration::readFiles] restoring file: 0018
[ INFO] [DepthCalibration::readFiles] restoring file: 0055_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0093_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0131_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0018_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0056
[ INFO] [DepthCalibration::readFiles] restoring file: 0094
[ INFO] [DepthCalibration::readFiles] restoring file: 0132
[ INFO] [DepthCalibration::readFiles] restoring file: 0019
[ INFO] [DepthCalibration::readFiles] restoring file: 0056_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0094_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0132_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0019_sync
[ INFO] [DepthCalibration::readFiles] restoring file: 0057
[ INFO] [DepthCalibration::readFiles] restoring file: 0095
[ INFO] [DepthCalibration::readFiles] restoring file: 0133
[ INFO] [DepthCalibration::readFiles] restoring file: 0020
[... 422 lines cut ...]
[ INFO] [DepthCalibration::calibrate] frame: ./0139_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0139_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0140_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0140_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0141_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0141_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0142_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0142_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0143_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0143_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0144_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0144_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0145_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0145_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0146_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0146_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0147_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0147_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0148_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0148_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0149_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0149_sync_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0150_depth.png
[ INFO] [DepthCalibration::calibrate] frame: ./0150_sync_depth.png
[ERROR] [DepthCalibration::compareDists] no distance sample data!

wiedemeyer, 2016-03-07:
Well, I don't know, why that happened. But the depth calibration is more a draft than a real feature. It is not mandatory.
But you could run it in a debugger and try to find out why no points get added to the `depthDists` and `imageDists` in `computePointDists`.

AlexxxBeck, 2016-03-07:
What is the command for running the depth calibration in a debugger?

AlexxxBeck, 2016-03-07:
Could it have something to do with this Error that I get everytime I recorded images? I kinda ignored this error since i always saw the image and I could record all images normally. This is the output that i get when I start recording:
alex@alex-UX303LAB:~/test$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record ir
[ INFO] [main] Start settings:
       Mode: record
     Source: ir
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] starting recorder...
[ INFO] [Recorder::startRecord] Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decrease min and max value for IR value rage
        [h] - increase min and max value for IR value rage
        [1] - decrease min value for IR value rage
        [2] - increase min value for IR value rage
        [3] - decrease max value for IR value rage
        [4] - increase max value for IR value rage
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1457370871868213993/compressed/set_parameters]
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1457370871868213993/compressed/set_parameters]

this error occurs for color, IR and sync.

wiedemeyer, 2016-03-07:
`gdb <catkin_ws>/devel/lib/kinect2_calibration/kinect2_calibration`
and with `set args ........` you can set the parameters. You can also use a graphical interface for gdb, like in qtcreator.

YoussefAlj, 2016-04-13:
I have some questions related to some calculus used during in DepthCalibration class.
1- in the function getPlane(const size_t index, cv::Mat &normal, double &distance): I don't understand which distance are computing in the line  distance = normal.dot(translation); is it the distance from the camera center to the plane?
2- in the function  computeDistance(const cv::Point &pointImage, const cv::Mat &normal, const double distance, cv::Mat point): 
what are you really computing in the line  double t = distance / normal.dot(point);?

Thank you 
Best regards

wiedemeyer, 2016-04-13:
1. is calculating the plane in Hesse normal form (normal and distance).
2. computeDistance computes the distance of a ray from the camera though a certain pixel to the plane.

wiedemeyer, 2016-04-14:
@AlexxxBeck any update?

AlexxxBeck, 2016-04-14:
Hello Thiemo,

I unfortunately couldn't resolve it but it worked for a friend of mine so I got the calibrated depth data and worked with it for a while now and everything works great.

closed 2016-04-14 by wiedemeyer