code-iai/iai_kinect2 #245: Running kinect2_bridge Headless

JustSpiffy, 2016-03-11:
I have tested my kinect2_bridge setup with a monitor and everything works fine. When I move my machine with the kinect and software to my platform, I use the machine without a monitor.

I ssh into my machine, and run roslaunch kinect2_bridge kinect2_bridge.launch but I get complaints from libfreenect2 that it can't create a glx window.

Is this an issue with how kinect2_bridge configures/uses libfreenect2? Or is it a libfreenect2 issue?

Errors below, thanks for any support...

libGL error: No matching fbConfigs or visuals found
libGL error: failed to log driver: swrast
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 GLX: Failed to create context: BadMatch (invalid parameter attributes)
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window

wiedemeyer, 2016-03-11:
Hi,
there are several options:
- build libfreenect2 with OpenCL (or CUDA) support and run `kinect2_bridge` with `depth_method:=opencl` (easiest and best solution)
- setup virtualgl to enable OpenGL over ssh
- use `depth_method:=cpu` (very slow)

JustSpiffy, 2016-03-11:
Thiemo,

Thanks so much for the blazing response! This is on an Intel NUC which seems to have limited support. I had to use the updated hardware stack (14.04.5) to even get ubuntu running on this beast, so I'm worried about OpenCL.

I have no need in tunneling OpenGL, is VirtualGL still suggested? I was looking into setting up a dummy display with `xserver-xorg-video-dummy`. Whats your opinion?

Constantin

wiedemeyer, 2016-03-11:
There is a ppa with Beignet 1.1.1 that should work out of the box.
https://github.com/code-iai/iai_kinect2#opencl-with-intel

JustSpiffy, 2016-03-11:
I went with your virtualgl suggestion, it seemed super easy cause I didn't want to mess with xorg.conf.

Since this machine will be on its own network with limited users, I installed virtualgl with no restrictions using: https://cdn.rawgit.com/VirtualGL/virtualgl/2.5/doc/index.html

After logging out and back in,

I ran 

`export DISPLAY=:0
roslaunch kinect2_bridge kinect2_bridge.launch`

closed 2016-03-15 by wiedemeyer