code-iai/iai_kinect2 #247: Running via SSH

soulslicer, 2016-03-18:
Good day,

I am using SSH to login to my machine and run kinect2_bridge. I am using Beignet OpenCL. It works fine when I login to ubuntu with the UI with X Server running and I run it. However, if I SSH in,

it throws an error saying "Packet Buffer is NULL" and /dev/dri/card0 is not authenticated. How can I get around this

wiedemeyer, 2016-03-29:
That is probably some permission issue. Did you connected with `ssh` or `ssh -X`? What happens if you run `./Protonect cl`? Or you could try `sudo ./Protonect cl`.

soulslicer, 2016-03-30:
Not a permission issue. Turns out you need X Server to Run if you are using Intel Beignet

closed 2016-03-30 by soulslicer

wiedemeyer, 2016-03-30:
From the [beignet documentation](https://www.freedesktop.org/wiki/Software/Beignet/):
*Normally, beignet needs to run under X server environment as normal user. If there isn't X server, beignet provides two alternative to run: * Run as root without X. * Enable the drm render nodes by passing drm.rnodes=1 to the kernel boot args, then you can run beignet with non-root and without X.*