code-iai/iai_kinect2 #248: Kinect2Bridge with Intel NUC 6th Gen

Pincello, 2016-03-18:
Hello i want to use a Kinect One on my NUC. When i launch kinect2_bridge i get this Output
With libfreenect2 only the kinect works fine.

`[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Error] [OpenCLDepthPacketProcessorImpl] program = cl::Program(context, source, &err): -34
[ INFO] [1458307235.710167036]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1458307235.710226989]: [DepthRegistration::New] Using OpenCL registration method!
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[ INFO] [1458307235.719831089]: [DepthRegistrationOpenCL::init] devices:
[ERROR] [1458307235.719857457]: [DepthRegistrationOpenCL::init] could not find any suitable device
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [1458307235.722153963]: [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[FATAL] [1458307235.827328863]: Service call failed!
^C[kinect2_points_xyzrgb_hd-6] killing on exit
[kinect2_points_xyzrgb_qhd-5] killing on exit
[kinect2_points_xyzrgb_sd-4] killing on exit
[kinect2-2] killing on exit
[kinect2_bridge-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
`

wiedemeyer, 2016-03-29:
There is some problem with your beignet installation:
```
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
```
And `./Protonect cl` is working?

mpatalberta, 2016-04-02:
We had problems with the nuc5i7 and had to modify the file cl_device_id.c  and cl_device_data.h.
You might try looking at that.

mpatalberta, 2016-04-02:
I have looked at the beignet 1.1.1 has the support for the 6th generation nuc. might try that at: https://01.org/beignet/downloads
build instructions:
https://www.freedesktop.org/wiki/Software/Beignet/

wiedemeyer, 2016-04-02:
@floe from the libfreenect2 developers has created a ppa with beignet 1.1.1 some time ago. You can install it following the instructions from the [README of libfreenect2](https://github.com/OpenKinect/libfreenect2#linux).

Pincello, 2016-04-05:
@wiedemeyer 
This is the Output from ./Protonect cl
Reinstalling the beignet 1.1.1 did not fix the Problem. The latest CPU Supportet is still 5 according to their Homepage.

Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
/dev/dri/card0 not authenticated
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Error] [OpenCLDepthPacketProcessorImpl] could not find any suitable device
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
libva info: VA-API version 0.38.1
libva info: va_getDriverName() returns 1
libva error: va_getDriverName() failed with operation failed,driver_name=i965
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): operation failed
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_READ_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 003154154647
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Error] [OpenCLDepthPacketProcessorImpl] OpenCLDepthPacketProcessor is not initialized!
[Error] [OpenCLDepthPacketProcessorImpl] OpenCLDepthPacketProcessor is not initialized!
[Error] [OpenCLDepthPacketProcessorImpl] OpenCLDepthPacketProcessor is not initialized!
[Debug] [Freenect2DeviceImpl] status 0x090000: 9775
[Debug] [Freenect2DeviceImpl] status 0x090000: 9775
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Error] [DepthPacketStreamParser] Packet buffer is NULL

[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Info] [Freenect2DeviceImpl] started
[Error] [DepthPacketStreamParser] Packet buffer is NULL

[Error] [DepthPacketStreamParser] Packet buffer is NULL
[device serial: 003154154647
device firmware: 2.3.3913.0
Error] [DepthPacketStreamParser] Packet buffer is NULL

[Error] [DepthPacketStreamParser] Packet buffer is NULL

[Error] [DepthPacketStreamParser] Packet buffer is NULL
GLFW error 65537 The GLFW library is not initialized
GLFW error 65537 The GLFW library is not initialized
GLFW error 65537 The GLFW library is not initialized
GLFW error 65537 The GLFW library is not initialized
GLFW error 65537 The GLFW library is not initialized
Failed to create opengl window.
[Error] [DepthPacketStreamParser] Packet buffer is NULL

[Error] [DepthPacketStreamParser] Packet buffer is NULL

mpatalberta, 2016-04-05:
I looked at the and I think I saw reference to the later releases.
Can you send me a link to the hardware you bought.
Pat

________________________________
- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

Pincello, 2016-04-06:
@mpatalberta 
This is my NUC : http://www.intel.com/content/www/us/en/nuc/nuc-kit-nuc6i5syk.html

mpatalberta, 2016-04-06:
You are correct not their.
You need to add it cl_device_data.h and cl_device_id.c  I had to do this for nuc7i5 before it was supported.
I think these are your chipset information that you will need.

122 CHIPSET(0x191B, skl_gt2, "Intel(R) HD Graphics 530 (Skylake GT2)")
  123 CHIPSET(0x191D, skl_gt2, "Intel(R) HD Graphics P530 (Skylake GT2)")
  124 CHIPSET(0x191E, skl_gt2, "Intel(R) HD Graphics 515 (Skylake GT2)")
  125 CHIPSET(0x1921, skl_gt2, "Intel(R) Skylake GT2")
  126 CHIPSET(0x1923, skl_gt3, "Intel(R) Iris Graphics 540 (Skylake GT3e)")
  127 CHIPSET(0x1926, skl_gt3, "Intel(R) HD Graphics 535 (Skylake GT3)")
  128 CHIPSET(0x1927, skl_gt3, "Intel(R) Iris Graphics 550 (Skylake GT3e)")
  129 CHIPSET(0x192A, skl_gt4, "Intel(R) Skylake GT4")
  130 CHIPSET(0x192B, skl_gt3, "Intel(R) Iris Graphics (Skylake GT3fe)")
  131 CHIPSET(0x1932, skl_gt4, "Intel(R) Skylake GT4")

Pincello, 2016-04-07:
@mpatalberta 
can you tell me where i find these Files ?

mpatalberta, 2016-04-07:
They are in Beignet/src

Pincello, 2016-04-20:
The issue was not beignet itself. I fixed it while i repeated the whole installation Process.  The first issue was during the OpenGL installation. The package there is for the wrong Ubuntu Version (ist vivid, should be trusty). The second one was in the OpenCL package. There was an error during the installation of the libra-dev......deb package. I installed the package by hand ( libva-dev_1.7.0-1_amd64.deb ) and now it works. Thanks Guys

CrumResearch, 2016-04-25:
Thank you very much for this discussion thread, it helped me make progress in installing iai_kinect2.  I also have to run kinect2_bridge as root, apparently.  Now my current issue involves a black IR image.  Maybe I should post a separate issue about this, Protonect shows the IR image fine but I bet a black image published to ROS and displayed with rviz.

closed 2016-06-03 by wiedemeyer

soulslicer, 2016-07-21:
Was this tested and working on the 6th Generation i5?

mabitter, 2016-12-14:
I had this issue as a result of launching over ssh.  Solution for me was to either work off the machine locally, or 'ssh -Y' in!  Hope that helps.

pallavbakshi, 2017-01-17:
I get the following error which keeps on repeating forever:
    [Error] [DepthPacketStreamParser] Packet buffer is NULL

This error is received when I run the Protonect example using
    ./Protonect cl
as mentioned on the website: [iai kinect 2](https://github.com/code-iai/iai_kinect2)

soulslicer, 2017-01-17:
Nah, the solution is just to do export DISPLAY=:0

BHAASKARR, 2017-02-24:
Hi All,

When i try to install ROS Indigo with libfreenect2 and iai_kinect2 in my computer with
  _Operating System: Ubuntu 14.04.5 LTS
  Kernel: Linux 4.4.0-64-generic_
I followed all steps mentioned in 
1. http://wiki.ros.org/indigo/Installation/Ubuntu
2. http://wiki.ros.org/catkin/Tutorials/create_a_workspace
3. https://github.com/openkinect/libfreenect2  
(Note: I haven't installed optional
 Example : _Install OpenCL (optional) which is not installed_)
**Protonect is working and showing color, depth and ir images as I got output when i 
Run the test program: ./bin/Protonect**
4. https://github.com/code-iai/iai_kinect2
I am not getting any output when i run 
_roslaunch kinect2_bridge kinect2_bridge.launch_
but i got following output.

[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Error] [OpenCLDepthPacketProcessorImpl] program = cl::Program(context, source, &err): -34
[ WARN] [1487965098.877783656]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1487965098.877814869]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1487965098.877830038]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1487965098.877847897]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1487965098.955753303]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1487965098.955800060]: [DepthRegistration::New] Using OpenCL registration method!
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[ INFO] [1487965098.967884971]: [DepthRegistrationOpenCL::init] devices:
[ERROR] [1487965098.967923702]: [DepthRegistrationOpenCL::init] could not find any suitable device
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[ERROR] [1487965098.968220168]: [Kinect2Bridge::start] Initialization failed!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[Error] [OpenCLAllocator] queue.enqueueUnmapMemObject(b->buffer, b->data, NULL, &event): -36
[FATAL] [1487965099.069387343]: Failed to load nodelet '/kinect2_bridge` of type `kinect2_bridge/kinect2_bridge_nodelet` to manager `kinect2'
^C[kinect2_points_xyzrgb_hd-6] killing on exit
[kinect2_points_xyzrgb_qhd-5] killing on exit
[kinect2_points_xyzrgb_sd-4] killing on exit
[kinect2_bridge-3] killing on exit
[kinect2-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Please help me.

Thanks,
Bhaskar.

SelvamArul, 2017-04-04:
@BHAASKARR Did manage to solve the issue? I am getting the same issue. In my case, the Protonect works fine (
[image] ) But I am not getting any depth topics (only depth topics. RGB is working fine). 

Description:    Ubuntu 14.04.5 LTS
4.4.0-62-generic

Log:

[ INFO] [1491315971.584686110]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/arul/arc/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Error] [CudaDepthPacketProcessorImpl] cudaGetDeviceCount(&deviceCount): unknown error
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:5 with serial 501544241942
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1491315971.659321073]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1491315971.659348892]: [Kinect2Bridge::initDevice]   0: 501544241942 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1491315971.757792519]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Info] [Freenect2DeviceImpl] started
[ INFO] [1491315972.077390476]: [Kinect2Bridge::initDevice] device serial: 501544241942
[ INFO] [1491315972.077435639]: [Kinect2Bridge::initDevice] device firmware: 2.3.3912.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL
[Error] [DepthPacketStreamParser] Packet buffer is NULL

mpatalberta, 2017-04-05:
Try the you mode this will tell you if the GPU is a problem or not.
Other problems could be:
 1- unsupported usb hardware
 2- the usb board is not supplying enough power
 3- you plugged the camera usb into a hub

[protonect]<https://urldefense.proofpoint.com/v2/url?u=https-3A__cloud.githubusercontent.com_assets_7311434_24662140_8b4c0504-2D1954-2D11e7-2D91e0-2D0c386056aac4.png&d=DwMFaQ&c=Qwsh1H-X9ypOoLLEcAIltQ&r=s3aTjjowuIOYrRraGZdX6w&m=oq0JDpgDD_K8HdG_2i2vgIJAkakwlQVi1o0fHxZkjpU&s=t-uUbFOmWXCzrti3GNmjrQNT92x19Vj6W-CAffXXA7U&e=> ) But I am not getting any depth topics (only depth topics. RGB is working fine).