code-iai/iai_kinect2 #250: kinect2_calib service already advertised

quentin1002, 2016-03-23:
Hi everybody, 

I am getting an error when running the record node for calibration. Here is the message I get: 

`lab@lab-Aspire-4750:~/kinect_cal_data$ rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color
[ INFO] [main] Start settings:
       Mode: record
     Source: color
      Board: chess
 Dimensions: 5 x 7
 Field size: 0.03
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] starting recorder...
[ INFO] [Recorder::startRecord] Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decrease min and max value for IR value rage
        [h] - increase min and max value for IR value rage
        [1] - decrease min value for IR value rage
        [2] - increase min value for IR value rage
        [3] - decrease max value for IR value rage
        [4] - increase max value for IR value rage
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1458721837266890808/compressed/set_parameters]
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1458721837266890808/compressed/set_parameters]
`
The image is published without any problem and the patterns recognised. Nevertheless I am not able to save the image and therefore not able to carry on with the calibration. 

For info, I run `rosrun kinect2_bridge kinect2_bridge _fps_limit:=2` in another terminal as indicated in the calibration tutorial. 

Am I doing anything wrong? Thanks in advance, 
Quentin

JimmyDaSilva, 2016-03-23:
Same error shows up for me, but the code runs and the images are saved anyway

quentin1002, 2016-03-24:
So you can still press space bar and save the image? 

Because for me, the node is running (is can see it in rqt) but the keyboard command are giving any results. Any advice?

JimmyDaSilva, 2016-03-24:
Yes I finished calibration yesterday.
I think you need to click on the image window, then press the keyboard.

wiedemeyer, 2016-03-29:
`Tried to advertise a service that is already advertised in this node` seems to be an issue related to image_transport plugins.

JamesChenxie, 2016-04-02:
Have you already solved this problem? I have this error too.

JamesChenxie, 2016-04-03:
I've already solved this problem. Just as Jimmy said, you need to click on the image window

closed 2016-04-18 by wiedemeyer

sakurazhu, 2016-09-28:
the same problem occured ,i have tried the way metioned by @JimmyDaSilva ,but  i can not save the image either. so can u give me a hand? .

JamesChenxie, 2016-10-18:
There must be something wrong with your chess board or you didn't choose the image window.（检查一下你的棋盘格板是不是按照标定标准设置的，如果标定程序已经检测到了你棋盘格板，选定标定显示窗口，按空格即可）

sakurazhu, 2016-10-19:
thank you . i finally fixed it ！

2016-10-19 

myworldpig

heyonghong, 2018-12-17:
@sakurazhu 请问一下你怎么解决这个问题的，我点了图像然后按空格或s，一点反应都没有。麻烦谢谢你啦

evalliETH, 2019-06-25:
I receive the same error but I can't click on the image window because there is none opening after typing in the code for calibration. Any ideas?

Eugenio

> I've already solved this problem. Just as Jimmy said, you need to click on the image window .