code-iai/iai_kinect2 #251: Point Cloud Compression, or publishing 32 bit cloud?

soulslicer, 2016-03-28:
Is there any such feature? The XYZ parameters are 64 bit by default and this causes bandwidth issues

wiedemeyer, 2016-03-29:
Point cloud coordinates are 32 bit float. If you have bandwidth problems, only subscribe to the compressed color and depth image and compute the point cloud where you need it. Or you can run the `point_cloud_xyzrgb` nodelet on that machine.

soulslicer, 2016-03-29:
Seems reasonable. I assumed the point cloud message type uses double. Just two further questions:

1. Is the point_cloud_xyzrgb nodelet able to read a compressed rgb stream to do the matching?

2.  point_cloud_xyzrgb doesnt seem to discard illogical points (clipped intensity or 0 intensity) and sends everything. perhaps it may be worth pushing some features to that nodelet

wiedemeyer, 2016-03-30:
1. It is not expensive.
2. You can republish the compressed images uncompressed on the other machine, [details here](http://wiki.ros.org/image_transport#republish)
3. It discards points that have zero depth. [see here](https://github.com/ros-perception/image_pipeline/blob/indigo/depth_image_proc/include/depth_image_proc/depth_traits.h#L48)

closed 2016-03-30 by soulslicer