code-iai/iai_kinect2 #252: Invalid argument error

kirstyellis, 2016-03-29:
I have installed libfreenect2 and iai_kinect2 as per the instructions given in your README. I can run Protonect and view the 4 images. However when I launch the kinect bridge and then try to run the viewer, I get the following:

 PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/kirsty/proj...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [10617]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a866e2ce-f59e-11e5-98f9-30b5c2021c29
process[rosout-1]: started with pid [10631]
started core service [/rosout]
process[kinect2-2]: started with pid [10642]
process[kinect2_bridge-3]: started with pid [10649]
process[kinect2_points_xyzrgb_sd-4]: started with pid [10650]
[ INFO] [1459249749.280929693]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_qhd-5]: started with pid [10651]
[ INFO] [1459249749.283599883]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
process[kinect2_points_xyzrgb_hd-6]: started with pid [10671]
[ INFO] [1459249749.333015730]: Initializing nodelet with 4 worker threads.
[ INFO] [1459249749.345983488]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1459249749.632090787]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/kirsty/projects/shadow_robot/base_deps/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics Haswell GT2 Desktop (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics Haswell GT2 Desktop (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 23 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @12:6 with serial 000610754947
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1459249749.816093904]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1459249749.816140327]: [Kinect2Bridge::initDevice]   0: 000610754947 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1459249749.910105788]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1459249750.532940105]: [Kinect2Bridge::initDevice] device serial: 000610754947
[ INFO] [1459249750.533015623]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
drm_intel_gem_bo_context_exec() failed: Invalid argument
[Error] [OpenCLDepthPacketProcessorImpl] enqueueNDRangeKernel failed: -5
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ WARN] [1459249750.919579315]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1459249750.919621537]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1459249750.919634952]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1459249750.919662076]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1459249750.971171973]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1459249750.971233121]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1459249750.981224308]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1459249750.981303870]: [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics Haswell GT2 Desktop (GPU)[Intel]
[ INFO] [1459249750.981334691]: [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics Haswell GT2 Desktop (GPU)[Intel]
[ INFO] [1459249751.077852555]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1459249751.077910140]: [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics Haswell GT2 Desktop (GPU)[Intel]
[ INFO] [1459249751.077924976]: [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics Haswell GT2 Desktop (GPU)[Intel]
[ INFO] [1459249751.170002647]: [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [1459249752.844532084]: [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
drm_intel_gem_bo_context_exec() failed: Invalid argument
[Error] [OpenCLDepthPacketProcessorImpl] enqueueNDRangeKernel failed: -5
drm_intel_gem_bo_context_exec() failed: Invalid argument
[Error] [OpenCLDepthPacketProcessorImpl] enqueueNDRangeKernel failed: -5
drm_intel_gem_bo_context_exec() failed: Invalid argument

I am using Ros Indigo and Ubuntu version 14.04 LTS.

I am launching roabridge with the following:

roslaunch kinect2_bridge kinect2_bridge.launch

and the viewer:

rosrun kinect2_viewer kinect2_viewer kinect2 sd cloud

which does not display any images or open a window.

There are no errors in the terminal where I am running the viewer.

I am also unable to calibrate the kinect as I get the same 'Invalid argument' error.

Any ideas why this isn't working?

kirstyellis, 2016-03-29:
Running  roslaunch kinect2_bridge kinect2_bridge.launch depth_method:=cpu reg_method:=cpu
has removed the error, closing.

closed 2016-03-29 by kirstyellis

wiedemeyer, 2016-03-29:
Yes, because then opencl is not used. There is probably something wrong with you beignet installation.