code-iai/iai_kinect2 #253: blank cloud viewer

hyderjake, 2016-03-29:
I have no errors, protonect works, and using the devel branch I'm saving the images with the images clearly visible in a folder, just the cloud viewer is blank. debug shows no issues with it running.

wiedemeyer, 2016-03-30:
Could you provide more information. What is the output from kinect2_bridge? Could you check that the depth images are not zero. Maybe provide some screenshots, etc.

hyderjake, 2016-03-30:
Here is the Kinect Bridge infor

[Info] [OpenCLDepthPacketProcessor] avg. time: 12.2019ms -> ~81.9546Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~10.7876ms (~92.6992Hz) publishing rate: ~28.9805Hz
[ INFO] [Kinect2Bridge::main] color processing: ~15.8375ms (~63.1414Hz) publishing rate: ~14.9899Hz
[Info] [DepthPacketStreamParser] 7 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 13.1492ms -> ~76.0502Hz
[ INFO] [Kinect2Bridge::main] depth processing: ~9.6386ms (~103.75Hz) publishing rate: ~27.3255Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.5287ms (~60.5008Hz) publishing rate: ~14.9957Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.8597ms -> ~55.9919Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~10.9686ms (~91.1696Hz) publishing rate: ~28.9932Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.3206ms (~61.2724Hz) publishing rate: ~14.9965Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 12.9351ms -> ~77.3092Hz
[Info] [DepthPacketStreamParser] 10 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~9.85521ms (~101.469Hz) publishing rate: ~25.9501Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.344ms (~61.1844Hz) publishing rate: ~14.9712Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 15.7583ms -> ~63.4588Hz
[Info] [DepthPacketStreamParser] 9 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~9.74822ms (~102.583Hz) publishing rate: ~26.9458Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.0147ms (~62.4428Hz) publishing rate: ~14.9699Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.8981ms -> ~55.8719Hz
[Info] [DepthPacketStreamParser] 5 packets were lost
[Info] [DepthPacketStreamParser] 6 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 15.7705ms -> ~63.4096Hz
[Info] [DepthPacketStreamParser] 7 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~9.32542ms (~107.234Hz) publishing rate: ~24.6229Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.9994ms (~58.8257Hz) publishing rate: ~14.9734Hz
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 16.4211ms -> ~60.8972Hz
[Info] [DepthPacketStreamParser] 5 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~10.1202ms (~98.8126Hz) publishing rate: ~26.615Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.0173ms (~62.4326Hz) publishing rate: ~14.9709Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.6046ms -> ~56.8033Hz
[Info] [DepthPacketStreamParser] 4 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~11.0682ms (~90.3492Hz) publishing rate: ~25.9783Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.325ms (~61.2558Hz) publishing rate: ~14.9875Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 15.9657ms -> ~62.6342Hz
[Info] [DepthPacketStreamParser] 10 packets were lost
[Info] [DepthPacketStreamParser] 6 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~11.0898ms (~90.1727Hz) publishing rate: ~26.9485Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.7041ms (~59.8654Hz) publishing rate: ~14.9714Hz
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 13.9735ms -> ~71.5643Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.835ms -> ~56.0696Hz
[Info] [DepthPacketStreamParser] 5 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~9.87041ms (~101.313Hz) publishing rate: ~27.5981Hz
[ INFO] [Kinect2Bridge::main] color processing: ~15.5876ms (~64.1534Hz) publishing rate: ~14.9628Hz
[Info] [DepthPacketStreamParser] 1 packets were lost

This is from the viewer

[ INFO] [main] topic color: /kinect2/qhd/image_color_rect
[ INFO] [main] topic depth: /kinect2/qhd/image_depth_rect
[ INFO] [main] starting receiver...
s[ INFO] [Receiver::saveCloudAndImages] saving cloud: ./0000_cloud.pcd
[ INFO] [Receiver::saveCloudAndImages] saving color: ./0000_color.jpg
[ INFO] [Receiver::saveCloudAndImages] saving depth: ./0000_depth.png
[ INFO] [Receiver::saveCloudAndImages] saving depth: ./0000_depth_colored.png
[ INFO] [Receiver::saveCloudAndImages] saving complete!

The saved images are clear and visible, I rotate the camera and nothing appears, gdb doesn't provide anything since it's not outputting any errors

hyderjake, 2016-03-30:
[image]
[image]
[image]

wiedemeyer, 2016-03-30:
Maybe it has something to do with your calibration. Please try to start it without your calibration. Just rename the calibration data folder to something else and restart the kinect2_bridge. Also include all the output of the kinect2_bridge from the beginning until `[ INFO] [Kinect2Bridge::main] waiting for clients to connect`.

hyderjake, 2016-03-30:
renamed the calibration and restarted roscore and kinect bridge

[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/jacob/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
NVIDIA: no NVIDIA devices found
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics 5600 BroadWell Mobile GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics 5600 BroadWell Mobile GT2 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:6 with serial 143374634247
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 143374634247 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 143374634247
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3912.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics 5600 BroadWell Mobile GT2 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics 5600 BroadWell Mobile GT2 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics 5600 BroadWell Mobile GT2 (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics 5600 BroadWell Mobile GT2 (GPU)[Intel]
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] 1 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~10.0259ms (~99.742Hz) publishing rate: ~28.9879Hz
[ INFO] [Kinect2Bridge::main] color processing: ~17.2738ms (~57.891Hz) publishing rate: ~26.3223Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 8.58726ms -> ~116.452Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.0035ms -> ~62.4862Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~9.8351ms (~101.677Hz) publishing rate: ~29.631Hz
[ INFO] [Kinect2Bridge::main] color processing: ~17.2569ms (~57.948Hz) publishing rate: ~29.9639Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 10.0959ms -> ~99.05Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.117ms -> ~62.0462Hz
[ INFO] [Kinect2Bridge::main] depth processing: ~10.0332ms (~99.6696Hz) publishing rate: ~29.9301Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.9046ms (~59.1556Hz) publishing rate: ~29.9301Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 8.5143ms -> ~117.449Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.7868ms -> ~63.3442Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[ INFO] [Kinect2Bridge::main] depth processing: ~8.98622ms (~111.281Hz) publishing rate: ~29.6332Hz
[ INFO] [Kinect2Bridge::main] color processing: ~16.6516ms (~60.0544Hz) publishing rate: ~29.9661Hz
[ INFO] [Kinect2Bridge::callbackStatus] no clients connected. stopping device...
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] disabling usb transfer submission...
[Info] [Freenect2DeviceImpl] canceling usb transfers...
[Info] [Freenect2DeviceImpl] stopped

wiedemeyer, 2016-03-30:
And do the images look the same again? `./Protonect cl` is working and showing a correct depth image?

hyderjake, 2016-03-30:
yes and yes, nothing seems off about the images individually, merging them seems to be the issue

wiedemeyer, 2016-03-30:
And what happens if you open the PCD file with `pcl_viewer`? Does `rosrun kinect2_viewer kinect2_viewer sd both` work?

hyderjake, 2016-03-30:
Image viewer was normal but blue, I think its supposed to be blue, cloud viewer was blank
pcl viewer had an error

> Loading 0000_cloud.pcd pcl_viewer: symbol lookup error: pcl_viewer: undefined symbol: _ZNK3pcl13visualization13PCLVisualizer16cameraFileLoadedEv

wiedemeyer, 2016-03-31:
maybe there be something wrong with your PCL installation? you could try to use rviz to visualize the point cloud using the DepthCloud plugin.

closed 2016-05-12 by wiedemeyer