code-iai/iai_kinect2 #254: Console only required for kinectv2 to run

mpatalberta, 2016-04-02:
The current configuration of the Kinectv2 usb configuration requires that the user has a graphical display. 
If you want to use the kinectv2 without the gui (X) for example. The following usb rule could be used:Ensure you can access the sensor
Give the sensor power and plug the usb connector from the sensor to your linux box. Verify you have the correct udev rules file to be able to access the sensor. You can verify by looking for the file “/etc/udev/rules.d/70-kinect2.rules”. It should contain the following:

BUS!="usb", SUBSYSTEM!=="usb_device", ACTION!="add", GOTO="kcontrol_rules_end"
# this file belongs in /etc/udev/rules.d/
# ATTR{product}=="Kinect2"
SUBSYSTEM=="usb", ATTRS{idVendor}==“045e", ATTRS{idProduct}=="02c4", MODE:="0666", OWNER:="root", GROUP:="video"
SUBSYSTEM=="usb", ATTRS{idVendor}==“045e", ATTRS{idProduct}=="02d8", MODE:="0666", OWNER:="root", GROUP:="video"
SUBSYSTEM=="usb", ATTRS{idVendor}==“045e", ATTRS{idProduct}=="02d9", MODE:="0666", OWNER:="root", GROUP:="video"

LABEL="kcontrol_rules_end"
If it doesn't exist, create it using sudo access and fill the contents with the text shown above.

The rules allow you to access the device if you are a member of the video group (or you are root of course). Make sure you are a member of the video group.

arwillis@visionlab16-pc:~/NRC/ROS_RGBD_Odometry/ros_ws/src$ cat /etc/group | grep video
video:x:44:arwillis
This shows my username “arwillis” next to the output line starting with “video” in the “/etc/groups” file. If you don't see your name next to the video group as above you must add yourself to the video group with the following command “sudo adduser <myusername> video”. Re-run the command above to verify you are now a member of the “video” group. If you are newly added to the “video” group, YOU MUST LOGOUT AND LOG BACK IN TO MAKE YOUR NEW GROUP ASSOCIATION/PERMISSIONS ACTIVE.

wiedemeyer, 2016-04-02:
Hi Patrick,
thanks for sharing those changes, but the `kinect2.rules` are part of the libfreenect2 driver. Any related issues could be better discussed on their issues page, or for changes a pull request to libfreenect2 could be created.
But how do the changes to the rule relate to running the Kinect2 without X?

mpatalberta, 2016-04-02:
Thiemo,
In my application we want to run the kinect2 without X.
I will move the suggestion to libfreenect2
Pat,

From: Thiemo Wiedemeyer <notifications@github.com<mailto:notifications@github.com>>
Reply-To: code-iai/iai_kinect2 <reply@reply.github.com<mailto:reply@reply.github.com>>
Date: Saturday, April 2, 2016 at 4:27 AM
To: code-iai/iai_kinect2 <iai_kinect2@noreply.github.com<mailto:iai_kinect2@noreply.github.com>>
Cc: "Brown, Patrick" <Patrick.Brown@zebra.com<mailto:Patrick.Brown@zebra.com>>
Subject: Re: [code-iai/iai_kinect2] Console only required for kinectv2 to run (#254)

r. Any related issues could be better discussed on their issues page, or for changes a pull request to libfreenect2 could be created.
But how do the changes to the rule relate to running the Kinect2 without X?

________________________________
- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

closed 2016-04-04 by wiedemeyer