code-iai/iai_kinect2 #255: Hardware Recommendation

lucbettaieb, 2016-04-05:
Hi @wiedemeyer ,

Is there any recommended hardware for using this driver?  I have tried running it on an GTX 540 and a 770 with little improvement in quality.  I lowered the quality and framerate params in the launch file and that helped a bit; but generally everything runs very very slowly.

Protonect runs very quickly, but whenever I run kinect2_bridge.launch and display the point cloud in RViz, I experience a big amount of lag.

Any suggestions?

Thanks!

wiedemeyer, 2016-04-05:
Hi,
it highly depends on what you want to do. For example, if you need high resolution point clouds or compression you need faster hardware. Some benchmarks are shown here:
https://github.com/OpenKinect/libfreenect2/wiki/Performance
An Intel Core-i processors with 4 cores and HD graphics should work fine. The Intel HD GPU can be used for decoding the Jpeg stream using VAAPI, and the depth images using OpenCL. To improve the OpenCL tasks dedicated GPUs can be used.
If it is slow for you, you need to check where the bottleneck lies. Most probably the CPU. Make sure to use OpenCL/CUDA and VAAPI, if possible, to lower the CPU load. Use `htop` to find out how high your CPU load is and what is using it. Rviz might also use a lot CPU, try the build in viewer, or just use `rostopic hz ...` to see how it performs. You can also try to use the DepthCloud plugin in RViz, maybe it works better than the PointCloud plugin.

lucbettaieb, 2016-04-05:
I'll try out your suggestions, thanks a lot!

closed 2016-04-05 by lucbettaieb