code-iai/iai_kinect2 #256: [Kinect2Bridge::initDevice] could not stop device

chenchengshao, 2016-04-05:
New to ROS, trying to install libfreenect2(Kinect2 works fine), and then iai_kinect2.
when I run roslaunch kinect2_bridge kinect2_bridge.launch
it reports [Kinect2Bridge::initDevice] could not stop device!, could you please help to look at this problem, any other information for you to trouble shooting.? 

> [ INFO] [1459897432.073721412]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
> [ INFO] [1459897432.073766513]: [Kinect2Bridge::initDevice]   0: 501185143042 (selected)
> [Info] [Freenect2DeviceImpl] opening...
> [Info] [Freenect2DeviceImpl] opened
> [ INFO] [1459897432.173501882]: [Kinect2Bridge::initDevice] starting kinect2
> [Info] [Freenect2DeviceImpl] starting...
> [Info] [Freenect2DeviceImpl] submitting rgb transfers...
> [Info] [Freenect2DeviceImpl] submitting depth transfers...
> [Info] [Freenect2DeviceImpl] started
> [ INFO] [1459897432.487934287]: [Kinect2Bridge::initDevice] device serial: 501185143042
> [ INFO] [1459897432.488004284]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
> [Info] [Freenect2DeviceImpl] stopping...
> [Info] [Freenect2DeviceImpl] canceling rgb transfers...
> [Info] [Freenect2DeviceImpl] canceling depth transfers...
> [ERROR] [1459897432.767493738]: [Kinect2Bridge::initDevice] could not stop device!
> [ERROR] [1459897432.767654569]: [Kinect2Bridge::start] Initialization failed!

wiedemeyer, 2016-04-08:
Are you using the latest libfreenect2 version? It could be that some communication with the device is not working correctly, please open an issue at libfreenect2 for this.

caselitz, 2016-04-08:
I have the same problem, using the latest libfreenect2 version. I will open an issue there...

closed 2016-04-14 by wiedemeyer