code-iai/iai_kinect2 #257: Possible error in Point Cloud Generation

soulslicer, 2016-04-10:
I took a look at your launch file, for your QHD Point Cloud example:

https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/launch/kinect2_bridge.launch

```
  <!-- qhd point cloud (960 x 540) -->
  <node pkg="nodelet" type="nodelet" name="$(arg base_name)_points_xyzrgb_qhd" machine="$(arg machine)"
        args="load depth_image_proc/point_cloud_xyzrgb $(arg nodelet_manager)" respawn="$(arg respawn)">
    <remap from="rgb/camera_info"             to="$(arg base_name)/qhd/camera_info"/>
    <remap from="rgb/image_rect_color"        to="$(arg base_name)/qhd/image_color_rect"/>
    <remap from="depth_registered/image_rect" to="$(arg base_name)/qhd/image_depth_rect"/>
    <remap from="depth_registered/points"     to="$(arg base_name)/qhd/points"/>
    <param name="queue_size" type="int" value="$(arg queue_size)"/>
  </node>
```

Here, you use /qhd/camera_info. However, the camera model being published here by your driver is not in the IR frame, rather it is in the RGB frame. I checked the values to be sure. This means you are generating the points where you consider the Depth Map to be in the RGB Frame.

Does this mean your iai_kinect code is publishing depth maps in the RGB Frame? How are you able to do this transform without first converting it to 3D point data.

wiedemeyer, 2016-04-11:
No, it's correct. The depth image is transformed to color frame. There are explanations in some closed issues already, for example here: #229

closed 2016-04-11 by wiedemeyer

soulslicer, 2016-04-11:
Noted thanks for the quick reply. I noticed you have an OpenCL of registration that transfers the depth map from the Depth/IR Frame into the RGB Frame.

Let me be clear that this is what you are doing but on the GPU:

1. You generate a set of 3D points using the IR Instrinsics and libfreenects depth image
2. You transform it into the RGB Frame
3. You project the points into this synthetic RGB image to get the depth image in the RGB Frame

This means that somewhere in your OpenCL Kernel, you have actually computed..the PointCloud?

wiedemeyer, 2016-04-13:
yes, the 3d points are computed in the registration process, it is done in both, the CPU and OpenCL implementation.