code-iai/iai_kinect2 #258: Performance test

giacomodabisias, 2016-04-11:
Hi Thiemo,
first great job with your grabber.

I finished developing a Ros grabber for the kinect2 based on my own grabber and I noticed some big performance differences between mine and yours. 
I could not find a reason for this differences so maybe you can help me with this issue.

Publishing fullhd registered rgb, depth (512x424) and the cloud at 30fps uses 340% cpu with your grabber while 110% with mine. 

It seems quite odd to me that the grabber alone consumes 340% cpu since it only has to grab the frames (construct a cloud) and publish them. Any ideas?

https://github.com/giacomodabisias/libfreenect2pclgrabber

Thanks

wiedemeyer, 2016-04-13:
Hi,
first of all, which topics are you subscribing to, and does your grabber produce the same output? For example which cloud are you referring too and which rgb stream are you using. Do you use the same libfreenect2 processors (VAAPI/TurboJPEG, OpenCL/CUDA) for both programs?

giacomodabisias, 2016-04-14:
Hi,
for the test I subscribed to the full HD rgb, the depth at 512x424 (there is a small difference since you convert the depth frame to 16bit while I leave it 32) and the cloud at 512x424. My grabber produces the same three topics. I tested it with CUDA, OpenGL and OpenCL.
Also the produced rgb and depth frames are registered.

wiedemeyer, 2016-04-14:
The main goal for the kinect2_bridge is to transmit images at full frame rate over gigabit ethernet, to be flexible and easy usable, low CPU usage is only a secondary goal. kinect2_bridge is doing some additional computations, which are probably the cause of the higher CPU load:
- all images are copied at the beginning, to free the buffers again, because processing could take long especially for compressed topics
- all images get flipped
- the color images are converted from BGRA to BGR, to save 25% bandwidth
- the depth images are converted to 16Bit integer instead of 32bit float to save an additional 50% bandwidth
- all `*_rect` images are also rectified using the calibration results
- the point clouds are created by nodelets from image_proc which also creates some overhead

giacomodabisias, 2016-04-14:
Ok thanks,
then that's probably it.

closed 2016-04-14 by giacomodabisias