code-iai/iai_kinect2 #259: Running libfreenect 2 on Intel NUC

pnambiar, 2016-04-12:
I'm trying to run kinect v2 with ROS. As a first step, I'm trying to run libfreenect 2 on Intel NUC -NUC5i7RYH with ubuntu Ubuntu 14.04.4 LTS.
I was able to install it successfully. However ./Protonect gives me the following error
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @3:3 with serial 503931742542
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65544 X11: RandR gamma ramp support seems broken
Xlib: extension "NV-GLX" missing on display ":0".
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.

Any leads will be highly appreciated

wiedemeyer, 2016-04-13:
Hi, did you followed [the instructions](https://github.com/OpenKinect/libfreenect2#linux)? Did you downloaded the debs for trusty with `libfreenect2/depends/download_debs_trusty.sh` and installed them?

pnambiar, 2016-04-13:
Yes

wiedemeyer, 2016-04-14:
you already have the same issue opened at libfreenect2, lets continue there and close this one.

closed 2016-04-14 by wiedemeyer