code-iai/iai_kinect2 #260: depth shift

YoussefAlj, 2016-04-14:
Hi, 

I did the IR and and depth calibration, I am wondering though why this depth shift between the depth provided by the kinect2 and and the computed plane exists?
Do you think there is a similar depth shift with Kinect1?

regards,

LovelyHorse, 2016-04-14:
If you're referring to a vertical difference between your base plane (as you may have in ROS) and that of your Kinect cloud, this could be related to noise below what your actual ground plane. If you mean your feed is flipped vertically (ground plane is aligned in the wrong direction), then a simple translation should see to it. This may look somethings like "0 0 *-pi / 2* 0 *-pi / 2* 0" though I've likely got the actual order wrong on that.

YoussefAlj, 2016-04-15:
I was referring to the shift in Z between the base plane and kinect2 depth measurements, using kinect2_calibration ros package. I applied the same code to the kinect1 and I found 5cm depth shift. Which sounds really huge to me. 
For the kinect1 IR calibration, I covered the IR projector with 2 post-it notes as explained in some ros tutorials, over-sampled  the ir images and equalized the histogram using @wiedemeyer code. And after calibration, I got a depth shift of 5cm. For the kinect2 using @wiedemeyer code, I have around 3cm depth shift.
cheers

wiedemeyer, 2016-04-15:
The depth calibration is a very preliminary piece of code. It just tries to lower the average error by subtracting it from the depth values. A much better solution would be to create a depth error model and apply it to the the depth data, but I didn't had time to implement something like that.
I never tried it on the first Kinect, but for the first Kinect there are tools for depth calibration available, for example this: http://www.ee.oulu.fi/~dherrera/kinect/
I guess you will get the best results if you use those tools.

YoussefAlj, 2016-04-15:
Thanks for the link. I think i will first try to compute the ground truth plane (that you computed in IR frame) in the color frame. I will get the depth registered frame to color frame, thanks to the depth_registration package. And then will compare the depth data with the ground truth plane (still in the color frame). And I'll see if this depth shift is still the same.

closed 2016-05-25 by wiedemeyer