code-iai/iai_kinect2 #261: [FATAL] [1460688984.161890434]: Service call failed! when I roslaunch  kinect2_bridge kinect2_bridge.launch.

JamesChenxie, 2016-04-15:
Hi everybody, 

I got an error when I roslaunch  kinect2_bridge kinect2_bridge.launch. 

Here is all the messages:

chenxieyuanli@chenxieyuanli-ThinkPad-T440p:~/catkin_ws$ roslaunch kinect2_bridge kinect2_bridge.launch 
... logging to /home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/roslaunch-chenxieyuanli-ThinkPad-T440p-20942.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://chenxieyuanli-ThinkPad-T440p:42375/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/chenxieyuan...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [20954]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a2c7fae6-02b5-11e6-96f2-5cc5d4b62664
process[rosout-1]: started with pid [20967]
started core service [/rosout]
process[kinect2-2]: started with pid [20979]
process[kinect2_bridge-3]: started with pid [20985]
process[kinect2_points_xyzrgb_sd-4]: started with pid [20986]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [20987]
[ INFO] [1460688983.090469007]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_hd-6]: started with pid [20996]
[ INFO] [1460688983.103552221]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1460688983.133257006]: Initializing nodelet with 8 worker threads.
[ INFO] [1460688983.145195176]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1460688983.210273291]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/chenxieyuanli/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 022766543547
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1460688983.352676849]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1460688983.352726862]: [Kinect2Bridge::initDevice]   0: 022766543547 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1460688983.484261625]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1460688983.830356069]: [Kinect2Bridge::initDevice] device serial: 022766543547
[ INFO] [1460688983.830418768]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[FATAL] [1460688984.161890434]: Service call failed!
[FATAL] [1460688984.161890351]: Service call failed!
[FATAL] [1460688984.163171368]: Service call failed!
[FATAL] [1460688984.163222797]: Service call failed!
[kinect2-2] process has died [pid 20979, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2-2.log].
log file: /home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2-2*.log
[kinect2_bridge-3] process has died [pid 20985, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2_bridge-3.log].
log file: /home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2_bridge-3*.log
[kinect2_points_xyzrgb_sd-4] process has died [pid 20986, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2_points_xyzrgb_sd-4.log].
log file: /home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2_points_xyzrgb_sd-4*.log
[kinect2_points_xyzrgb_qhd-5] process has died [pid 20987, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2_points_xyzrgb_qhd-5.log].
log file: /home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2_points_xyzrgb_qhd-5*.log
[kinect2_points_xyzrgb_hd-6] process has died [pid 20996, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2_points_xyzrgb_hd-6.log].
log file: /home/chenxieyuanli/.ros/log/a2c7fae6-02b5-11e6-96f2-5cc5d4b62664/kinect2_points_xyzrgb_hd-6*.log
[kinect2_bridge-3] restarting process
process[kinect2_bridge-3]: started with pid [21136]
[kinect2_points_xyzrgb_sd-4] restarting process
process[kinect2_points_xyzrgb_sd-4]: started with pid [21137]
[kinect2_points_xyzrgb_qhd-5] restarting process
process[kinect2_points_xyzrgb_qhd-5]: started with pid [21138]
[kinect2_points_xyzrgb_hd-6] restarting process
process[kinect2_points_xyzrgb_hd-6]: started with pid [21139]
[ INFO] [1460688984.469607751]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1460688984.473193139]: waitForService: Service [/kinect2/load_nodelet] could not connect to host [chenxieyuanli-ThinkPad-T440p:54824], waiting...

Hope to your reply, thanks.

wiedemeyer, 2016-04-15:
try `rosrun kinect2_bridge kinect2_bridge` and see if that works fine.

JamesChenxie, 2016-04-15:
Thanks for your reply, but here comes a new one as follow:

**_[ERROR] [1460711798.290781614]: [Kinect2Bridge::receiveIrDepth] received invalid frame format_**

Here are the details
... logging to /home/chenxieyuanli/.ros/log/be311300-02ea-11e6-86bc-5cc5d4b62664/roslaunch-chenxieyuanli-ThinkPad-T440p-3790.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://chenxieyuanli-ThinkPad-T440p:35611/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/chenxieyuan...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: cpu
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [3802]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to be311300-02ea-11e6-86bc-5cc5d4b62664
process[rosout-1]: started with pid [3815]
started core service [/rosout]
process[kinect2-2]: started with pid [3828]
process[kinect2_bridge-3]: started with pid [3833]
process[kinect2_points_xyzrgb_sd-4]: started with pid [3834]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [3835]
[ INFO] [1460711792.396782139]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_hd-6]: started with pid [3846]
[ INFO] [1460711792.402696815]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1460711792.434501697]: Initializing nodelet with 8 worker threads.
[ INFO] [1460711792.444315445]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1460711792.601592654]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/chenxieyuanli/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: cpu
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 022766543547
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1460711792.741674047]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1460711792.741725597]: [Kinect2Bridge::initDevice]   0: 022766543547 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1460711792.873517071]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1460711793.217336668]: [Kinect2Bridge::initDevice] device serial: 022766543547
[ INFO] [1460711793.217388351]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ INFO] [1460711793.499648212]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1460711793.499725274]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1460711793.568134625]: [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [1460711797.896101020]: [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ERROR] [1460711798.290781614]: [Kinect2Bridge::receiveIrDepth] received invalid frame format
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] -2 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.3411ms -> ~69.7294Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.4468ms -> ~54.2101Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.275ms -> ~70.0527Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.3672ms -> ~57.5797Hz

wiedemeyer, 2016-04-15:
make sure your libfreenect2 is up to date.

JamesChenxie, 2016-04-15:
I've reinstall libfreenect2 for many times using 'git clone https://github.com/OpenKinect/libfreenect2.git'.

But still failed.

I didn't install cuda. Should  I do that?
 
Here are the details:
... logging to /home/chenxieyuanli/.ros/log/e5341cbc-0313-11e6-9e5c-5cc5d4b62664/roslaunch-chenxieyuanli-ThinkPad-T440p-6051.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://chenxieyuanli-ThinkPad-T440p:47449/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/chenxieyuan...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: cpu
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6063]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e5341cbc-0313-11e6-9e5c-5cc5d4b62664
process[rosout-1]: started with pid [6076]
started core service [/rosout]
process[kinect2-2]: started with pid [6085]
process[kinect2_bridge-3]: started with pid [6094]
process[kinect2_points_xyzrgb_sd-4]: started with pid [6095]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [6096]
[ INFO] [1460729467.225720914]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_hd-6]: started with pid [6105]
[ INFO] [1460729467.235470340]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1460729467.269638790]: Initializing nodelet with 8 worker threads.
[ INFO] [1460729467.277131076]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1460729467.346926562]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/chenxieyuanli/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: cpu
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 022766543547
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1460729467.490725350]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1460729467.490757865]: [Kinect2Bridge::initDevice]   0: 022766543547 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1460729467.624397113]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1460729467.968128976]: [Kinect2Bridge::initDevice] device serial: 022766543547
[ INFO] [1460729467.968162454]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ INFO] [1460729468.250936507]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1460729468.251008069]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1460729468.318521257]: [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [1460729472.228134126]: [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ERROR] [1460729472.638904567]: [Kinect2Bridge::receiveIrDepth] failure in depth packet processor from libfreenect2
[Info] [DepthPacketStreamParser] -2 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 15.4965ms -> ~64.5309Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.2526ms -> ~70.1624Hz

JamesChenxie, 2016-04-15:
Here are the messages when I run**_ ./bin/Protonect_**

[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 022766543547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 022766543547
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Haswell Mobile - 1.7.0
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] 5 packets were lost
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing 

Here are the messages when I run**_ NiViewer2_**
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 022766543547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 022766543547
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Haswell Mobile - 1.7.0
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] 5 packets were lost
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

wiedemeyer, 2016-04-15:
There seems to be something wrong with the ir/depth frames you receive, the error messages are pretty clear and there is something not working right in your libfreenect2 installation. Did you ran `make install` after updating libfreenect2? Check that `kinect2_bridge` links to the right `libfreenect2.so` by using `ldd`.

Please add the following code to [line 944 in kinect2_bridge.cpp](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/src/kinect2_bridge.cpp#L944), to print the status and format of both frames:
```
OUT_INFO("ir status: " << irFrame->status << " format: " << irFrame->format);
OUT_INFO("depth status: " << depthFrame->status << " format: " << depthFrame->format);
```

JamesChenxie, 2016-04-16:
Thanks Thiemo! It works now!!

Stupid I (face palm) still don't know what's going wrong.

Failed so many times, and I reinstall Ubuntu and ROS. That's done (face palm again.)

closed 2016-04-16 by wiedemeyer