code-iai/iai_kinect2 #262: kinect2_bridge using opencl? but switches to opengl when client connected.

hanshammel1337, 2016-04-15:
hello,

when i start `kinect2_bridge.launch` everything seems fine. opencl chooses my dedicated graphic card and the bridge is waiting for clients:
```
ben@ben-laptop:~/libfreenect2/build/bin$ roslaunch kinect2_bridge kinect2_bridge.launch 
... logging to /home/ben/.ros/log/4d754ebe-02f5-11e6-9763-ac811248baa3/roslaunch-ben-laptop-6562.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ben-laptop:39118/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/ben/catkin_...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6574]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4d754ebe-02f5-11e6-9763-ac811248baa3
process[rosout-1]: started with pid [6587]
started core service [/rosout]
process[kinect2-2]: started with pid [6602]
process[kinect2_bridge-3]: started with pid [6605]
process[kinect2_points_xyzrgb_sd-4]: started with pid [6606]
[ INFO] [1460716327.713504237]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_qhd-5]: started with pid [6607]
process[kinect2_points_xyzrgb_hd-6]: started with pid [6620]
[ INFO] [1460716327.724265773]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1460716327.752445131]: Initializing nodelet with 4 worker threads.
[ INFO] [1460716327.770637638]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1460716328.032658979]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ben/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 000000000
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1460716330.700342172]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1460716330.700463133]: [Kinect2Bridge::initDevice]   0: 000000000 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1460716330.797604728]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1460716331.132894109]: [Kinect2Bridge::initDevice] device serial: 000000000
[ INFO] [1460716331.132944647]: [Kinect2Bridge::initDevice] device firmware: 2.3.3913.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [1460716331.373685615]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1460716331.373783457]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1460716331.373871037]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1460716331.374002613]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1460716331.493728019]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1460716331.493855746]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1460716331.686222773]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1460716331.686304399]: [DepthRegistrationOpenCL::init]   0: Turks (GPU)[Advanced Micro Devices, Inc.]
[ INFO] [1460716331.686333763]: [DepthRegistrationOpenCL::init]   1: Intel(R) Core(TM) i5-2410M CPU @ 2.30GHz (CPU)[GenuineIntel]
[ INFO] [1460716331.686368373]: [DepthRegistrationOpenCL::init] selected device: Turks (GPU)[Advanced Micro Devices, Inc.]
[ INFO] [1460716331.832333485]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1460716331.832410500]: [DepthRegistrationOpenCL::init]   0: Turks (GPU)[Advanced Micro Devices, Inc.]
[ INFO] [1460716331.832451810]: [DepthRegistrationOpenCL::init]   1: Intel(R) Core(TM) i5-2410M CPU @ 2.30GHz (CPU)[GenuineIntel]
[ INFO] [1460716331.832489430]: [DepthRegistrationOpenCL::init] selected device: Turks (GPU)[Advanced Micro Devices, Inc.]
[ INFO] [1460716331.997848938]: [Kinect2Bridge::main] waiting for clients to connect
```
when i run the example for the viewer `rosrun kinect2_viewer kinect2_viewer sd cloud` the bridge output shows, that now openGL is used. the viewer windows is just black.
```
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] -2 packets were lost
[ INFO] [1460716745.353498318]: [Kinect2Bridge::main] depth processing: ~11.442ms (~87.397Hz) publishing rate: ~29.6352Hz
[ INFO] [1460716745.353587121]: [Kinect2Bridge::main] color processing: ~3.06391ms (~326.381Hz) publishing rate: ~17.6479Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 12.0005ms -> ~83.3298Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 32.8789ms -> ~30.4146Hz
[ INFO] [1460716748.353881401]: [Kinect2Bridge::main] depth processing: ~10.3305ms (~96.8007Hz) publishing rate: ~29.9961Hz
[ INFO] [1460716748.353980375]: [Kinect2Bridge::main] color processing: ~3.24659ms (~308.016Hz) publishing rate: ~19.3308Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 11.3232ms -> ~88.314Hz
```
`depth_method` and `reg_method` are by default set to `opencl`. when i force this arguments to be opencl in the launchfile, kinect2_bridge start fails with:
```
...
[ INFO] [1460717026.790307537]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/ben/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[ERROR] [1460717026.790443675]: [Kinect2Bridge::initPipeline] OpenCL depth processing is not available!
[ERROR] [1460717026.790475563]: [Kinect2Bridge::start] Initialization failed!
[FATAL] [1460717026.871391937]: Service call failed!
```
i can't see any depth images with integrated viewer or with rviz :( please help me. any suggestions?
why openCL seesm to be fine when using `default` and fails when using `opencl`?

wiedemeyer, 2016-04-15:
You have to compile libfreenect2 with OpenCL support. Your current libfreenect does not have OpenCL enabled. I guess you installed OpenCL after compiling libfreenect2, since kinect2_bridge detected OpenCL. Remove your libfreenect2 installation, remove the libfreenect2 build folder and run `cmake` and `make install` again. `make` should output that it is compiling `opencl_depth_processor.cpp`. Then rebuild kinect2_bridge.

hanshammel1337, 2016-04-18:
I thought it was compiled with OpenCL suppport...
Nevertheless I reinstalled my graphic card driver and made a fresh libfreenect2 build; now everything is fine with kinect2_bridge.

```
-- Feature list:
--   CUDA    no
--   CXX11    yes
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    no
--   TegraJPEG    no
--   Threading    std::thread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
```
and
```
[ 84%] Building CXX object CMakeFiles/freenect2.dir/src/opengl_depth_packet_processor.cpp.o
[ 88%] Building CXX object CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o
Linking CXX shared library lib/libfreenect2.so
```

closed 2016-04-18 by hanshammel1337