code-iai/iai_kinect2 #264: HD Depth

pnambiar, 2016-04-15:
I managed to get kinect 2 working with iai_kinect2. I just noticed that while the HD color images are being published at around 30 hz, HD depth is only being published at 5 hz. Is there a way around it?

wiedemeyer, 2016-04-15:
Which depth processor is libfreenect2 using? Try OpenCL/CUDA/OpenGL.
Please post the terminal output from kinect2_bridge here.

pnambiar, 2016-04-15:
OPENGL

narayanan@narayanan-desktop:~$ roslaunch kinect2_bridge kinect2_bridge.launch
... logging to /home/narayanan/.ros/log/53ae0618-0327-11e6-ad0d-3413e82c92ff/roslaunch-narayanan-desktop-15014.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://narayanan-desktop:41518/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/narayanan/c...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [15026]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 53ae0618-0327-11e6-ad0d-3413e82c92ff
process[rosout-1]: started with pid [15039]
started core service [/rosout]
process[kinect2-2]: started with pid [15056]
process[kinect2_bridge-3]: started with pid [15057]
process[kinect2_points_xyzrgb_sd-4]: started with pid [15058]
[ INFO] [1460737812.887180206]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_qhd-5]: started with pid [15059]
[ INFO] [1460737812.892400497]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
process[kinect2_points_xyzrgb_hd-6]: started with pid [15079]
[ INFO] [1460737812.908407976]: Initializing nodelet with 4 worker threads.
[ INFO] [1460737812.914141243]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1460737813.054147390]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/narayanan/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Broadwell - 1.7.0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 503931742542
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1460737813.231100623]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1460737813.231143793]: [Kinect2Bridge::initDevice]   0: 503931742542 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1460737813.328746737]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1460737813.646130748]: [Kinect2Bridge::initDevice] device serial: 503931742542
[ INFO] [1460737813.646160481]: [Kinect2Bridge::initDevice] device firmware: 2.3.3912.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [1460737813.886257689]: [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [1460737813.886282406]: [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [1460737813.886298313]: [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [1460737813.886317133]: [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [1460737813.927804315]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1460737813.928365995]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1460737813.993180415]: [Kinect2Bridge::main] waiting for clients to connect

wiedemeyer, 2016-04-15:
Install Beignet and use OpenCL for depth registration.

pnambiar, 2016-04-15:
Works. Thanks a bunch!!!

pnambiar, 2016-04-15:
Also the calibration seems to be off. Is this the procedure that needs to be followed?
https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration

wiedemeyer, 2016-04-16:
Yes, you need to calibrate it first, but that it should be registered fine.
You can use ExactTime.

closed 2016-04-18 by wiedemeyer

pnambiar, 2016-04-18:
Is ExactTime same as this procedure?
https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration