code-iai/iai_kinect2 #265: Problems with cmake Ubuntu 14.04

aaramirezd, 2016-04-15:
Hi everybody,

I got an error with: 

catkin_make -DCMAKE_BUILD_TYPE="Release" 

I Install libfreenect2 repeatedly and Protonect and this works
    ./Protonect gl to test OpenGL support.
    ./Protonect cl to test OpenCL support.
    ./Protonect cpu to test CPU support.

I need help, I'm trying to work with Kinect v2 and ROS to make gaze 3d tracking

below the error report

root@andres:/home/andres/catkin_ws# catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/andres/catkin_ws
Source space: /home/andres/catkin_ws/src
Build space: /home/andres/catkin_ws/build
Devel space: /home/andres/catkin_ws/devel
Install space: /home/andres/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/andres/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/andres/catkin_ws/build"
####
[ 42%] Built target kinect2_registration
Linking CXX executable /home/andres/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
[ 57%] Built target kinect2_calibration
[ 71%] Built target kinect2_bridge_nodelet
[ 85%] Built target kinect2_viewer
/usr/bin/ld: warning: libcudart.so.7.5, needed by /home/andres/freenect2/lib/libfreenect2.so, not found (try using -rpath or -rpath-link)
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaMemcpyToSymbolAsync' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaDeviceSynchronize' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaFreeHost' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaHostAlloc' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `__cudaUnregisterFatBinary' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `__cudaRegisterFatBinary' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaGetLastError' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaMalloc' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaMemcpy' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `__cudaRegisterVar' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaMemcpyAsync' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaGetDeviceProperties' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaConfigureCall' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaSetDevice' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaLaunch' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaGetErrorString' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaGetDeviceCount' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `__cudaRegisterFunction' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaFree' sin definir
/home/andres/freenect2/lib/libfreenect2.so: referencia a `cudaSetupArgument' sin definir
collect2: error: ld returned 1 exit status
make[2]: *** [/home/andres/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

wiedemeyer, 2016-04-18:
It seems to be a problem of your system configuration. Make sure that `libcudart.so.7.5` is found. For example add the folder to `LD_LIBRARY_PATH`.

aaramirezd, 2016-04-18:
Ok, I can build the package, thanks. 
Now I have a problem with libva, look in the above issues and can not find any related.

below the error report

... logging to /home/andres/.ros/log/04421702-0568-11e6-917d-408d5c706ebe/roslaunch-andres-6895.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://andres:37691/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/andres/catk...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: jade
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6907]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 04421702-0568-11e6-917d-408d5c706ebe
process[rosout-1]: started with pid [6920]
started core service [/rosout]
process[kinect2-2]: started with pid [6931]
process[kinect2_bridge-3]: started with pid [6937]
process[kinect2_points_xyzrgb_sd-4]: started with pid [6938]
[ INFO] [1460985499.260067861]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_qhd-5]: started with pid [6940]
[ INFO] [1460985499.268920024]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
process[kinect2_points_xyzrgb_hd-6]: started with pid [6957]
[ INFO] [1460985499.275552162]: Initializing nodelet with 4 worker threads.
[ INFO] [1460985499.289873251]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1460985499.439341691]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/andres/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
[... 49 lines cut ...]
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[ INFO] [1460985500.220423253]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1460985500.220549583]: [DepthRegistrationOpenCL::init]   0: GeForce GT 730 (GPU)[NVIDIA Corporation]
[ INFO] [1460985500.220631226]: [DepthRegistrationOpenCL::init] selected device: GeForce GT 730 (GPU)[NVIDIA Corporation]
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[ INFO] [1460985500.262937989]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1460985500.263039980]: [DepthRegistrationOpenCL::init]   0: GeForce GT 730 (GPU)[NVIDIA Corporation]
[ INFO] [1460985500.263125706]: [DepthRegistrationOpenCL::init] selected device: GeForce GT 730 (GPU)[NVIDIA Corporation]
[ INFO] [1460985500.316627342]: [Kinect2Bridge::main] waiting for clients to connect

Thank you very much

closed 2016-04-19 by wiedemeyer