code-iai/iai_kinect2 #267: color image OK, but IR image is black

CrumResearch, 2016-04-25:
Anyone else encounter the case where Protonect shows all four images just fine, but the ROS IR image from kinect2_bridge /kinect2/sd/image_ir is black as shown by rviz and when using kinect2_calibration to record "ir" images?

wiedemeyer, 2016-04-25:
what is the output on the terminal?

CrumResearch, 2016-04-25:
Hi Thiemo!  Do you think this error is significant, and do you have any suggestion on how to eliminate it?
Failed to release test userptr object! (9) i915 kernel driver may not be sane!

Here is the output from kinect2_bridge:

root@dgk1:~/catkin_ws/devel/lib/kinect2_bridge# rosrun kinect2_bridge kinect2_bridge _fps_limit:=2
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: 2
       calib_path: /home/setup/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) HD Graphics Haswell Ultrabook GT3 reserved (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) HD Graphics Haswell Ultrabook GT3 reserved (GPU)[Intel]
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Haswell - 1.7.0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 500444441942
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 500444441942 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 500444441942
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [DepthRegistration::New] Using OpenCL registration method!
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics Haswell Ultrabook GT3 reserved (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics Haswell Ultrabook GT3 reserved (GPU)[Intel]
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
[ INFO] [DepthRegistrationOpenCL::init] devices:
[ INFO] [DepthRegistrationOpenCL::init]   0: Intel(R) HD Graphics Haswell Ultrabook GT3 reserved (GPU)[Intel]
[ INFO] [DepthRegistrationOpenCL::init] selected device: Intel(R) HD Graphics Haswell Ultrabook GT3 reserved (GPU)[Intel]
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
[ INFO] [Kinect2Bridge::main] waiting for clients to connect

wiedemeyer, 2016-04-25:
I never saw this message before. What is the output once you subscribe to a topic? `./Protonect cl` is working correctly, right? Or just `./Protonect`? You could also try to use OpenGL and see if that gives depth images.

CrumResearch, 2016-04-25:
Yes, "./Protonect cl" displays a good image, not all black.  Protonect without arguments displays all 4 good images.

Output is all 0s from the ROS topic /kinect2/sd/image_ir .  I have also used rviz to display the IR image as well and depth images, with only 0s.

setup@dgk1:~/catkin_ws/devel$ rostopic echo /kinect2/sd/image_ir | head -20 | cut -c 1-80
header:
  seq: 53
  stamp:
    secs: 1461612457
    nsecs: 605618540
  frame_id: kinect2_ir_optical_frame
height: 424
width: 512
encoding: 16UC1
is_bigendian: 0
step: 1024
data: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 ...
---
header:
  seq: 54
  stamp:
    secs: 1461612458
    nsecs: 108965146
  frame_id: kinect2_ir_optical_frame
height: 424

CrumResearch, 2016-04-26:
Hi Thiemo and all, this issue is looking like an OpenCL / Beignet issue, since Protonect with the "-cl" option display only one of the four images unlike Protonect with no arguments which shows all four images.  And, beignet utests do not succeed.  This is on a Mac Mini computer, Model A1347 EMC 2840.

I'll post more later but I will be out of the office today so it may be tomorrow when I continue working this issue.  Here is the output of beignet utests_run, maybe I will need to dig into beignet and perhaps use a discussion forum for that implementation of OpenCL.  I wonder about the error message "Failed to release test userptr object! (9) i915 kernel driver may not be sane!", it looks like a kernel or library version/bugfix issue from what info I have found so far.

 

root@dgk1:~/beignet/Beignet-1.0.2-Source/build/utests# ./utest_run
platform number 1
platform_profile "FULL_PROFILE"
platform_name "Intel Gen OCL Driver"
platform_vendor "Intel"
platform_version "OpenCL 1.2 beignet 1.0.2"
platform_extensions "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_icd"
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
device_profile "FULL_PROFILE"
device_name "Intel(R) HD Graphics Haswell Ultrabook GT3 reserved"
device_vendor "Intel"
device_version "OpenCL 1.2 beignet 1.0.2"
device_extensions "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_icd"
device_opencl_c_version "OpenCL C 1.2 beignet 1.0.2"
Failed to release test userptr object! (9) i915 kernel driver may not be sane!
21 image formats are supported
[CL_R CL_UNORM_INT8]
[CL_R CL_UNORM_INT16]
[CL_R CL_SIGNED_INT8]
[CL_R CL_SIGNED_INT16]
[CL_R CL_SIGNED_INT32]
[CL_R CL_UNSIGNED_INT8]
[CL_R CL_UNSIGNED_INT16]
[CL_R CL_UNSIGNED_INT32]
[CL_R CL_HALF_FLOAT]
[CL_R CL_FLOAT]
[CL_RGBA CL_UNORM_INT8]
[CL_RGBA CL_UNORM_INT16]
[CL_RGBA CL_SIGNED_INT8]
[CL_RGBA CL_SIGNED_INT16]
[CL_RGBA CL_SIGNED_INT32]
[CL_RGBA CL_UNSIGNED_INT8]
[CL_RGBA CL_UNSIGNED_INT16]
[CL_RGBA CL_UNSIGNED_INT32]
[CL_RGBA CL_HALF_FLOAT]
[CL_RGBA CL_FLOAT]
[CL_BGRA CL_UNORM_INT8]
test_load_program_from_bin_file()Failed: set OCL_KERNEL_PATH. This is where the kiss kernels are (-1)

summary:
----------
  total: 710
  run: 1
  pass: 0
  fail: 0
  pass rate: 0.000000
root@dgk1:~/beignet/Beignet-1.0.2-Source/build/utests#

root@dgk1:~/beignet/Beignet-1.0.2-Source/build/utests# uname -a
Linux dgk1 3.16.0-30-generic #40~14.04.1-Ubuntu SMP Thu Jan 15 17:43:14 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
root@dgk1:~/beignet/Beignet-1.0.2-Source/build/utests# cat /etc/*release*
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=14.04
DISTRIB_CODENAME=trusty
DISTRIB_DESCRIPTION="Ubuntu 14.04.4 LTS"
NAME="Ubuntu"
VERSION="14.04.4 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.4 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
root@dgk1:~/beignet/Beignet-1.0.2-Source/build/utests#

wiedemeyer, 2016-04-28:
Until the OpenCL problems are fixed, you could use the OpenGL implementation for the depth processing and the CPU depth registration in kinect2_bridge. https://github.com/code-iai/iai_kinect2/tree/master/kinect2_bridge#usage

closed 2016-05-25 by wiedemeyer