code-iai/iai_kinect2 #268: Compiling problem

portaloffreedom, 2016-04-26:
I had a strange error compiling this package
```cmake
-- Found OpenCL: /usr/lib64/libOpenCL.so (found version "2.0") 
-- CPU based depth registration enabled
-- OpenCL based depth registration enabled
-- Performing Test OpenCL_C_WORKS
CMake Error at /usr/share/cmake-3.5/Modules/CheckCSourceCompiles.cmake:75 (try_compile):
  Unknown extension ".c" for file

    /tmp/yaourt-tmp/aur-ros-jade-kinect2-registration/src/build/CMakeFiles/CMakeTmp/src.c

  try_compile() works only for enabled languages.  Currently these are:

    CXX

  See project() command to enable other languages.
Call Stack (most recent call first):
  cmake/CheckOpenCLICDLoader.cmake:6 (CHECK_C_SOURCE_COMPILES)
  CMakeLists.txt:56 (include)

-- Performing Test OpenCL_C_WORKS - Failed
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
-- Configuring incomplete, errors occurred!
```

I fixed it by adding this line to the cmake file
```cmake
ENABLE_LANGUAGE(C)
```

wiedemeyer, 2016-04-28:
Maybe the cause is that in the CMakeLists.txt in the second line only `CXX` is mentioned. Does it work if you just add `C` or you remove anything expect the project name?

portaloffreedom, 2016-04-29:
I think is because of my build environment configuration. If I use `source /opt/ros/jade/setup.bash` is giving that error, if I use `/opt/ros/jade/env.sh cmake ...` is working correclty.

portaloffreedom, 2016-04-29:
I suppose this is a distrubution specific thing, or an error in ros-catkin. You can close the issue, at least if someone else has the same problem, you can quickly refer them here ;)

closed 2016-05-04 by wiedemeyer