code-iai/iai_kinect2 #270: Calibration no Pictures are Taken

arminkazim7, 2016-04-29:
First of all i cannot use roslaunch to start the programms so i start them by hand via ./kinect2_bridge 
and ./kinect2_calibration
First of all i get:
[ INFO] [main] Start settings:
       Mode: record
     Source: sync
      Board: chess
 Dimensions: 7 x 6
 Field size: 0.108
Dist. model: 5 coefficients
Topic color: /kinect2/hd/image_mono
   Topic ir: /kinect2/sd/image_ir
Topic depth: /kinect2/sd/image_depth
       Path: ./

[ INFO] [main] starting recorder...
[ INFO] [Recorder::startRecord] Controls:
   [ESC, q] - Exit
 [SPACE, s] - Save current frame
        [l] - decrease min and max value for IR value rage
        [h] - increase min and max value for IR value rage
        [1] - decrease min value for IR value rage
        [2] - increase min value for IR value rage
        [3] - decrease max value for IR value rage
        [4] - increase max value for IR value rage
After that i get the error:
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1461939741226382238/compressed/set_parameters]
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1461939741226382238/compressed/set_parameters]
in the terminal where i am running kinect2_calibration and nothing else happens and i cant take any photos.
I would really like to calibrate my Ir sensor so any help would be appreciated.

arminkazim7, 2016-04-29:
The ./kinect2_bridge Terminal gives this output meanwhile:
[Info] [OpenGLDepthPacketProcessor] avg. time: 19.0594ms -> ~52.4676Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [VaapiRgbPacketProcessor] avg. time: 5.71542ms -> ~174.965Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 19.1296ms -> ~52.2749Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [VaapiRgbPacketProcessor] avg. time: 5.68178ms -> ~176.001Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 19.2756ms -> ~51.8789Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [VaapiRgbPacketProcessor] avg. time: 5.63175ms -> ~177.565Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 19.4578ms -> ~51.3933Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 5.46991ms -> ~182.818Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 18.4429ms -> ~54.2215Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 5.30099ms -> ~188.644Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 18.2033ms -> ~54.935Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 5.40757ms -> ~184.926Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 18.4342ms -> ~54.2471Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 5.40323ms -> ~185.074Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 18.6919ms -> ~53.4992Hz

arminkazim7, 2016-05-02:
By the way i am using Ubuntu 14.10 with ROS JADE installed.

wiedemeyer, 2016-05-04:
If you cant use roslaunch something with you setup is wrong.
Does the calibration display images on the screen? If so, you have to place the calibration pattern in the view, as soon as it gets detected (colored overlay of the corners is show) you can record data.

arminkazim7, 2016-05-04:
The calibration displays no image at all.
While ./Protonect of libfreenect does.

arminkazim7, 2016-05-04:
In Advance Thanks for your Help

wiedemeyer, 2016-05-09:
could you check if there is data published, for example `rostopic hz /kinect2/hd/image_mono`, and the same for `/kinect2/sd/image_ir` and `/kinect2/sd/image_depth`.

arminkazim7, 2016-05-09:
so i ran ./kinect2_bridge 
and after that rostopic hz /kinect2/sd/image_ir
i get the output in the terminal where rostopic runs 
subscribed to [/kinect2/sd/image_ir]
but nothing else
while in the terminal where ./kinect2_bridge runs i get stuff like
Info] [VaapiRgbPacketProcessor] avg. time: 5.30598ms -> ~188.467Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 18.6802ms -> ~53.5326Hz
also when i run in another terminal ./kinect2_calibration 
i still get no output in the rostopic hz Terminal

wiedemeyer, 2016-05-10:
Maybe there is something wrong with you ros installation.
Anyway, what is the complete terminal output of kinect2_bridge when you run it?

arminkazim7, 2016-05-10:
linaro@linaro-ESPRIMO-P720:~/cat*/devel/lib/kinect2_bridge$ ./kinect2_bridge
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/linaro/cat*/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opengl
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
libva info: VA-API version 0.35.1
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_35
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Haswell Desktop - 1.3.2
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 005273445247
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 005273445247 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 005273445247
[ INFO] [Kinect2Bridge::initDevice] device firmware: 2.3.3913.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using CPU registration method!
[ INFO] [DepthRegistration::New] Using CPU registration method!
[ INFO] [Kinect2Bridge::main] waiting for clients to connect

arminkazim7, 2016-05-10:
after starting ./kinect2_calibration i will receive in kinect2_bridge Terminal
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ERROR] [Kinect2Bridge::receiveIrDepth] received invalid frame format
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] -2 packets were lost
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 19.0686ms -> ~52.4421Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 5.63468ms -> ~177.472Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 18.5122ms -> ~54.0184Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 5.67141ms -> ~176.323Hz

wiedemeyer, 2016-05-10:
`[ERROR] [Kinect2Bridge::receiveIrDepth] received invalid frame format` this means that something with the frames you received from libfreenect2 is wrong. Maybe it is related to you opengl driver. Try to run `rosrun kinect2_bridge kinect2_bridge _depth_method:=cpu`.

arminkazim7, 2016-05-10:
Running: linaro@linaro-ESPRIMO-P720:~/cat/devel/lib/kinect2_bridge$ ./kinect2_bridge _depth_method:=cpu
gives me the output:
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/linaro/cat*/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
libva info: VA-API version 0.35.1
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_35
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Haswell Desktop - 1.3.2
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 005273445247
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 005273445247 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 005273445247
[ INFO] [Kinect2Bridge::initDevice] device firmware: 2.3.3913.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for color intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using sensor defaults for ir intrinsic parameters.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for rotation and translation.
[ WARN] [Kinect2Bridge::initCalibration] using defaults for depth shift.
[ INFO] [DepthRegistration::New] Using CPU registration method!
[ INFO] [DepthRegistration::New] Using CPU registration method!
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ERROR] [Kinect2Bridge::receiveIrDepth] received invalid frame format
[Info] [DepthPacketStreamParser] 30 packets were lost
[Info] [DepthPacketStreamParser] 30 packets were lost
[Info] [DepthPacketStreamParser] 30 packets were lost

wiedemeyer, 2016-05-11:
make sure your libfreenect2 version is up to date, that you did `make install` and that kinect2_bridge links against the right version with `ldd`.

arminkazim7, 2016-05-25:
okay i got everything running using ubuntu 14.04 and new installation following the guide
also i had to install the created package as superuser by
source devel/setup.bash
otherwise i couldnt access the kinect

arminkazim7, 2016-05-25:
Here is my complete guide 
//install ros
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-get update
sudo apt-get install ros-indigo-desktop-full
sudo apt-get install ros-indigo-ros-base
sudo rosdep init
rosdep update
echo "source /opt/ros/indigo/setup.bash" >> ~/.bashrc
source ~/.bashrc
source /opt/ros/indigo/setup.bash
sudo apt-get install python-rosinstall
sudo apt-get install cmake python-catkin-pkg python-empy python-nose python-setuptools libgtest-dev build-essential
//install catkin
Download
https://github.com/ros/catkin
cd catkin
sudo python setup.py install
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src
catkin_init_workspace
//libfreenect2
git clone https://github.com/OpenKinect/libfreenect2.git
cd depends; ./download_debs_trusty.sh
sudo apt-get install build-essential cmake pkg-config
sudo dpkg -i debs/libusb*deb
sudo apt-get install libturbojpeg libjpeg-turbo8-dev
sudo apt-add-repository ppa:floe/beignet; sudo apt-get update; sudo apt-get install beignet-dev; sudo dpkg -i debs/ocl-icd*deb
sudo dpkg -i debs/{libva,i965}*deb; sudo apt-get install -f
sudo apt-add-repository ppa:deb-rob/ros-trusty && sudo apt-get update
sudo apt-get install libopenni2-dev
cd lib*
mkdir build && cd build
cmake .. -DENABLE_CXX11=ON
make
make install
//iai
git clone https://github.com/code-iai/iai_kinect2.git
cd iai_kinect2
rosdep install -r --from-paths .
cd ~/catkin_ws
catkin_make -DCMAKE_BUILD_TYPE="Release"
cd devel
chmod 777 setup.bash
./setup.bash
sudo su source devel/setup.bash
//After that you can use rosrun and roslaunch
rosrun kinect2_bridge kinect2_bridge _fps_limit:=2
//and then run the calibration where pressing the store button is only available if chessboard is detected!
rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color

closed 2016-05-26 by wiedemeyer