code-iai/iai_kinect2 #272: no chessboard seen on ir-image

royalheadwear, 2016-05-03:
Hi,
I followed the instructions for calibrating the kinect2 sensor.

The recording of colorimages worked fine, but when I tried to take a picture in the ir-mode the chessboard was not detected (added a screenshot) . Is it possible, that this happens because I didn't print it with an laser printer?

[image]

arminkazim7, 2016-05-03:
maybe its too far away have you tried moving the board a little closer?

royalheadwear, 2016-05-03:
Okay... Thanks to the issue #80 I found out, that it actually was the ink printed Pattern. I printed it with a laser printer now and now its far better!!!

closed 2016-05-03 by royalheadwear