code-iai/iai_kinect2 #275: error: ‘_mm_mfence’ was not declared in this scope      inline void fence() { _mm_mfence(); }

Bremiat, 2016-05-06:
Hi,

we were trying to get iai_kinect running in order to use a Kinect V2 by following the instructions on the website, but we encountered several problems.

What we managed to do:
- Install Ubuntu 14.04 (Kubuntu)
- Install ROS Indigo
- Setup ROS Environment
- Download libfreenect2

And here are the problems:

The first encountered problem was that libfreenect could not be built with -DENABLE_CXX11=ON
`cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 -DENABLE_CXX11=ON`
This command came back with the error attached in the file "Fehler Freenect mit CXX11.txt"

Without the CXX1 flag, libfreenect could be built and Protonect was successfully running. However, the iai_kinect package could not be built. It gave the following error:
/usr/include/CL/cl.hpp: In function ‘void cl::detail::fence()’:
/usr/include/CL/cl.hpp:1041:38: error: ‘_mm_mfence’ was not declared in this scope
     inline void fence() { _mm_mfence(); }
Details can be found in the attached file "Fehler beim Bauen von IAI_Kinect.txt"

We suspect that it has something to do with the fact the libfreenect2 was not built with the CXX11 flag. As the first error of libfreenect2 appeared after the file opencl_depth_packet_processor.cpp, we suspected that it may have something to do with CL.

Running clinfo gives the following error:
drm_intel_gem_bo_context_exec() failed: Invalid argument
Beignet: self-test failed: (3, 7, 5) + (5, 7, 3) returned (3, 7, 5)
This can usually be fixed by upgrading Linux to >= 4.2,
see /usr/share/doc/beignet-opencl-icd/Beignet.mdwn.gz or http://www.freedesktop.org/wiki/Software/Beignet/
Beignet: warning - disable atomic in L3 feature.
Beignet: disabling non-working device
E: -1

Another probably unrelated problem is that the rules for Kinect permissions could not be copied. The following command fails:
`sudo cp libfreenect2/rules/90-kinect2.rules /etc/udev/rules.d/`
cp: der Aufruf von stat für »libfreenect2/rules/90-kinect2.rules“ ist nicht möglich: Datei oder Verzeichnis nicht gefunden

For better understanding of what we did so far, we also attached the bash history to this message.

Does anyone encounter a similar problem? Any idea how to solve it?

Thank you very much in advance!
The IAT Bremen team

[Bash History.txt](https://github.com/code-iai/iai_kinect2/files/252044/Bash.History.txt)
[Fehler beim Bauen von IAI_Kinect.txt](https://github.com/code-iai/iai_kinect2/files/252045/Fehler.beim.Bauen.von.IAI_Kinect.txt)
[Fehler clinfo.txt](https://github.com/code-iai/iai_kinect2/files/252046/Fehler.clinfo.txt)
[Fehler Freenect mit CXX11.txt](https://github.com/code-iai/iai_kinect2/files/252048/Fehler.Freenect.mit.CXX11.txt)
[Fehler udev rules.txt](https://github.com/code-iai/iai_kinect2/files/252047/Fehler.udev.rules.txt)

wiedemeyer, 2016-05-09:
Regarding the CXX11 error: If you don't need openni, disable it with cmake (or ccmake), enable CXX11 and try again. BTW: which CPU and Kernel are you using?

The udev file location chnaged in libfreenect2, the instruction on the site is outdated, but the one from the libfreenect2 instructions are correct: `sudo cp <libfreenect2>/platform/linux/udev/90-kinect2.rules /etc/udev/rules.d/`

Bremiat, 2016-05-09:
we use I7 Intel und Kubuntu 14.04

wiedemeyer, 2016-05-10:
You can get the kernel version with `uname -a` and the CPU name with `grep "model name" /proc/cpuinfo`. Did it worked with CXX11 enabled and openni disabled?

Bremiat, 2016-05-10:
short Info:
I had installed the 32bit version of Ubuntu.
Yesterday  I restalled the System with 64bit version of Ubuntu. The Infomation you see is now from the newest  version.
Also I used the CXX11 enabled flag and now it is runing.

here are the Information:

Linux ros 3.19.0-59-generic #65~14.04.1-Ubuntu SMP Tue Apr 19 18:57:09 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
ros@ros:/usr/share/man/de$ grep "model name" /proc/cpuinfo
model name      : Intel(R) Core(TM) i7-4771 CPU @ 3.50GHz
model name      : Intel(R) Core(TM) i7-4771 CPU @ 3.50GHz
model name      : Intel(R) Core(TM) i7-4771 CPU @ 3.50GHz
model name      : Intel(R) Core(TM) i7-4771 CPU @ 3.50GHz
model name      : Intel(R) Core(TM) i7-4771 CPU @ 3.50GHz
model name      : Intel(R) Core(TM) i7-4771 CPU @ 3.50GHz
model name      : Intel(R) Core(TM) i7-4771 CPU @ 3.50GHz
model name      : Intel(R) Core(TM) i7-4771 CPU @ 3.50GHz

Bis: code-iai/iai_kinect2

wiedemeyer, 2016-05-10:
So the problems are solved? Or are there still any open issues?

Bremiat, 2016-05-10:
Thanks for the help.
I think the problem was the 32 bit operating system.
Unfortunately, I had read no note about it in advance.
Up here, I have no questions more, but maybe when calibrating.

Many thanks for your effort

closed 2016-05-11 by wiedemeyer