code-iai/iai_kinect2 #276: Calibration reprojection error

nomi321, 2016-05-07:
for the sync (combination of ir and color extrinsics), the re-projection error I get is around 6. is it fine ? How can i reduce this further. initially it was found to be about 20 .

wiedemeyer, 2016-05-09:
There are some hints at [Calibrating the Kinect One](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration#calibrating-the-kinect-one) and in issue #66.
Results > 1 are not good, try to get something < 0.3.

nomi321, 2016-05-10:
I brought this to 1 and now again after taking more samples it again jumped from 1 to 4. what is the accurate way of taking samples. i simply click on ir window and then press space and then chose color window and press space. is this correct way ?

nomi321, 2016-05-10:
and secondly for seprate ir and color calibration the error appears to be less then 0.3. why do we need this sync and for what purpose ? can i leave it

nomi321, 2016-05-10:
[ INFO] [CameraCalibration::calibrateExtrinsics] Camera Matrix Color:
[1300.14062044615, 0, 985.5373788706105;
  0, 1279.780680642094, 432.4111577673459;
  0, 0, 1]

[ INFO] [CameraCalibration::calibrateExtrinsics] Camera Matrix Ir:
[380.5669915619917, 0, 255.2172299071115;
  0, 375.3668586320726, 190.9437118587339;
  0, 0, 1]

[ INFO] [CameraCalibration::calibrateExtrinsics] calibrating Color and Ir extrinsics...
[ INFO] [CameraCalibration::calibrateExtrinsics] re-projection error: 4.45373

[ INFO] [CameraCalibration::calibrateExtrinsics] Rotation:
[0.9998699148649068, 0.003768411537235434, 0.0156828703536662;
  -0.004436473505762782, 0.9990745046054982, 0.04278378139097957;
  -0.01550712903398099, -0.04284779249581588, 0.9989612583215423]
[ INFO] [CameraCalibration::calibrateExtrinsics] Translation:
[-0.06704198812959684; 0.01175268700770214; 0.1431620097586768]
[ INFO] [CameraCalibration::calibrateExtrinsics] Essential:
[0.0004528840294016934, -0.1435330906721518, 0.00561546687286486;
  0.1421037577486768, -0.002333107826615507, 0.06921754006093558;
  -0.01145372815371442, -0.06702423003965799, -0.003052625630802329]
[ INFO] [CameraCalibration::calibrateExtrinsics] Fundamental:
[-7.479007108940614e-08, 2.403168196478256e-05, -0.004922528937197104;
  -2.384060542193034e-05, 3.968457369221618e-07, 0.001589401996774928;
  0.01284185383612433, -0.009265754962794578, 1]

wiedemeyer, 2016-05-10:
It will record both images at once, it does not matter which window is active, as long as one of them is active.
It is only needed for registering the depth image to the color image, if you don't need a depth image in the color frame or high resolution point clouds, than you can skip it.
Your calibration results for color and ir are totally bad. Consider redoing the calibration and please read and follow the hints at [Calibrating the Kinect One](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration#calibrating-the-kinect-one) and in issue #66 carefully.

nomi321, 2016-05-10:
reprojection error was less than 0.2 then how can you say that these results are bad. How can one check that results are satisfactory or not ?

wiedemeyer, 2016-05-10:
Because the values in the camera matrix are different than normal. Could it be that you only took a couple of images? If you took 100 with different positions and rotations of the pattern and you re-projection error is low, than it is probably a good calibration, if you only took 5 images it doesn't mean anything.

nomi321, 2016-05-10:
i took about 80-90 images for each .. what should be the values ? I am going to do it again but i was only focusing reprojection error and i was happy that it was below 0.3 for IR and Color. Kindly let me know the values

nomi321, 2016-05-10:
please let me know the reasonable values for the three cases. I shall be thankful to you . I have trying this since week and I am not successful and even fed up . Please help

nomi321, 2016-05-10:
is this fine now ? it is after taking about 100 images with different orientations

[ INFO] [CameraCalibration::calibrateIntrinsics] re-projection error: 0.392941

[ INFO] [CameraCalibration::calibrateIntrinsics] Camera Matrix:
[1103.649648530384, 0, 994.9722500886552;
  0, 1103.406365127304, 539.2101977041719;
  0, 0, 1]
[ INFO] [CameraCalibration::calibrateIntrinsics] Distortion Coeeficients:
[-0.0282658834407647, 0.02217597448866161, 0.00545302487800403, -0.005926335439252119, -0.03086045976164221]

wiedemeyer, 2016-05-10:
not perfect but looks better. Good results can be found here https://github.com/code-iai/iai_kinect2/tree/master/kinect2_bridge/data
Make sure that you have images from different distances all over the image area and also from the corners. Pan, tilt and rotate the pattern on all axis. And make sure, that the dimensions of the pattern are exact.

JimmyDaSilva, 2016-05-10:
@nomi321 Also make sure the printed pattern is really flat on your surface

nomi321, 2016-05-10:
@wiedemeyer what if i cant achive 1060 as in my case after alot of struggle it reached upto 1100. How it will effect ?

closed 2016-06-14 by wiedemeyer