code-iai/iai_kinect2 #277: cpu usage

pnambiar, 2016-05-10:
Kinect2 is taking up a large portion of the cpu and slowing down my processing program. Any solution?

wiedemeyer, 2016-05-11:
Use OpenCL/CUDA for depth processing, VAAPI for JPEG decoding, OpenCL for registration and use compressed topics only in case of bandwidth limitations.
On my system it uses 25% CPU on an i7-4770K, when subscribing to `/kinect2/hd/image_color_rect` and `/kinect2/hd/image_depth_rect` in parallel and processing frames at 30Hz.

pnambiar, 2016-05-11:
I think I'm using OPENCL and VAAPI.  Its using up 250 % of the intel NUC when I'm streaming /kinect2/hd/image_color_rect and /kinect2/hd/image_depth_rect in parallel. I guess the only option now is to used the compressed images?

core service [/rosout] found
process[kinect2-1]: started with pid [15026]
process[kinect2_bridge-2]: started with pid [15027]
process[kinect2_points_xyzrgb_sd-3]: started with pid [15028]
[ INFO] [1462973676.460045681]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1462973676.464715479]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
process[kinect2_points_xyzrgb_qhd-4]: started with pid [15046]
process[kinect2_points_xyzrgb_hd-5]: started with pid [15049]
[ INFO] [1462973676.490201883]: Initializing nodelet with 4 worker threads.
[ INFO] [1462973676.509448574]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1462973676.755441110]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/narayanan/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Broadwell - 1.7.0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 503931742542
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1462973677.196238519]: [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [1462973677.196276732]: [Kinect2Bridge::initDevice]   0: 503931742542 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [1462973677.292073652]: [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [1462973677.610118045]: [Kinect2Bridge::initDevice] device serial: 503931742542
[ INFO] [1462973677.610170400]: [Kinect2Bridge::initDevice] device firmware: 2.3.3912.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ INFO] [1462973678.172381832]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1462973678.172433472]: [DepthRegistration::New] Using OpenCL registration method!
[ INFO] [1462973678.180994992]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1462973678.181044379]: [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [1462973678.181070565]: [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [1462973678.278096295]: [DepthRegistrationOpenCL::init] devices:
[ INFO] [1462973678.278136373]: [DepthRegistrationOpenCL::init]   0: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [1462973678.278150738]: [DepthRegistrationOpenCL::init] selected device: Intel(R) Iris Graphics 6100 BroadWell U-Processor GT3 (GPU)[Intel]
[ INFO] [1462973678.375157320]: [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[ INFO] [1462973678.379719026]: [Kinect2Bridge::main] waiting for clients to connect
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] -2 packets were lost
[Info] [DepthPacketStreamParser] 3 packets were lost
[ INFO] [1462973681.445476700]: [Kinect2Bridge::main] depth processing: ~76.5944ms (~13.0558Hz) publishing rate: ~21.3104Hz
[ INFO] [1462973681.447383850]: [Kinect2Bridge::main] color processing: ~96.8059ms (~10.3299Hz) publishing rate: ~11.6542Hz
[Info] [DepthPacketStreamParser] 4 packets were lost
[Info] [OpenCLDepthPacketProcessor] avg. time: 15.9134ms -> ~62.84Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 12.7242ms -> ~78.5904Hz
[ INFO] [1462973684.450706075]: [Kinect2Bridge::main] depth processing: ~0.903891ms (~1106.33Hz) publishing rate: ~30.6129Hz
[ INFO] [1462973684.450898139]: [Kinect2Bridge::main] color processing: ~25.7245ms (~38.8735Hz) publishing rate: ~30.2801Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 7.38692ms -> ~135.374Hz
[ INFO] [1462973686.034389857]: [Kinect2Bridge::callbackStatus] no clients conne

wiedemeyer, 2016-05-11:
I think this is the best you can get from a low power dual core CPU.
If you don't need 30 fps, just limit them to save CPU, or use stronger hardware like a quad core i7 destop CPU. You can also do your processing on a different machine and dedicate the NUC to processing the Kinect2 data. Using compressed images will only consume more CPU.

closed 2016-05-13 by wiedemeyer