code-iai/iai_kinect2 #279: Two problems for using opencl

JamesChenxie, 2016-05-12:
**Hi everybody, 

I got some troubles when I ran kinect2 with opencl

The first one, I used opencl as reg_method::**

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/chenxieyuan...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: cpu
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: False
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_device: -1
 * /kinect2_bridge/reg_method: opencl
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [17634]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 25db7f44-17e1-11e6-a106-5cc5d4b62664
process[rosout-1]: started with pid [17647]
started core service [/rosout]
process[kinect2-2]: started with pid [17659]
process[kinect2_bridge-3]: started with pid [17665]
process[kinect2_points_xyzrgb_sd-4]: started with pid [17666]
process[kinect2_points_xyzrgb_qhd-5]: started with pid [17667]
process[kinect2_points_xyzrgb_hd-6]: started with pid [17673]
[ INFO] [1463016645.622453410]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
[ INFO] [1463016645.627384363]: waitForService: Service [/kinect2/load_nodelet] has not been advertised, waiting...
[ INFO] [1463016645.659869801]: Initializing nodelet with 8 worker threads.
[ INFO] [1463016645.669024472]: waitForService: Service [/kinect2/load_nodelet] is now available.
[ INFO] [1463016645.741641571]: [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/chenxieyuanli/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: cpu
     depth_device: -1
       reg_method: opencl
[... 181 lines cut ...]
[ INFO] [1463017291.902429855]: [Kinect2Bridge::initDevice] device serial: 022766543547
[ INFO] [1463017291.902474913]: [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
**_drm_intel_gem_bo_context_exec() failed: Invalid argument
[Error] [OpenCLDepthPacketProcessorImpl]_** queue.enqueueNDRangeKernel(kernel_processPixelStage1, cl::NullRange, cl::NDRange(IMAGE_SIZE), cl::NullRange, &eventWrite, &eventPPS1[0]): -5
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[ INFO] [1463017292.467325711]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1463017292.467469410]: [DepthRegistration::New] Using CPU registration method!
[ INFO] [1463017292.536850169]: [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [1463017308.023987560]: [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
**[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!**
[Info] [DepthPacketStreamParser] -2 packets were lost
[ INFO] [1463017311.360507492]: [Kinect2Bridge::main] depth processing: ~infms (~0Hz) publishing rate: ~0Hz
[ INFO] [1463017311.360578124]: [Kinect2Bridge::main] color processing: ~15.6279ms (~63.9879Hz) publishing rate: ~26.2584Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 5.7987ms -> ~172.453Hz

Hope to your reply, thanks.

wiedemeyer, 2016-05-12:
Hi,
there is something wrong with you OpenCL installation. Which OS/Kernel and Beignet version are you using?

JamesChenxie, 2016-05-12:
Thanks Thiemo! It's so nice of you to help me so many times. I've already solved these problems.

Here is the status of my computer:
Intel® Core™ i7-4710MQ CPU @ 2.50GHz × 8 
Intel® Haswell Mobile & GeForce GT 730M/PCIe/SSE2

chenxieyuanli@chenxieyuanli-ThinkPad-T440p:~$ uname -r
3.16.0-70-generic

I download "intel_code_builder_for_opencl_2015_ubuntu_5.0.0.43_x64.tgz" from https://software.intel.com/en-us/articles/intel-code-builder-for-opencl-api
The following steps worked for me on Ubuntu 15.04 based on Noah's instructions above:

		$ sudo apt-get install -y rpm alien libnuma1

		$ tar -xvf intel_code_builder_for_opencl_2015_ubuntu_5.0.0.43_x64.tgz
		$ cd intel_code_builder_for_opencl_2015_ubuntu_5.0.0.43_x64

		$ sudo rpm --import PUBLIC_KEY.PUB

		$ cd rpm

		$ fakeroot alien --to-deb opencl-1.2-base-5.0.0.43-1.x86_64.rpm
		$ fakeroot alien --to-deb opencl-1.2-intel-cpu-5.0.0.43-1.x86_64.rpm

		$ sudo dpkg -i opencl-1.2-base_5.0.0.43-2_amd64.deb
		$ sudo dpkg -i opencl-1.2-intel-cpu_5.0.0.43-2_amd64.deb

		The above installs the library files and installable client driver registration in /opt/intel/opencl-1.2-5.0.0.43.
		Two more steps were needed to run an OpenCL program.

		Add library to search path:
		$ sudo nano /etc/ld.so.conf.d/intelOpenCL.conf
		Add the line:
		/opt/intel/opencl-1.2-5.0.0.43/lib64

		Link to the intel icd file in the expected location:
		$ sudo ln /opt/intel/opencl-1.2-5.0.0.43/etc/intel64.icd /etc/OpenCL/vendors/intel64.icd
	        $ sudo ldconfig
		$ sudo clinfo

And that's done! I can use kinect2 with  qhd in 30Hz. that's amazing!

Thank you Thiemo!

Number of platforms:				 3
  Platform Profile:				 FULL_PROFILE
  Platform Version:				 OpenCL 1.2 beignet 1.1.1
  Platform Name:				 Intel Gen OCL Driver
  Platform Vendor:				 Intel
  Platform Extensions:				 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_spir cl_khr_icd
  Platform Profile:				 FULL_PROFILE
  Platform Version:				 OpenCL 1.2 LINUX
  Platform Name:				 Intel(R) OpenCL
  Platform Vendor:				 Intel(R) Corporation
  Platform Extensions:				 cl_khr_icd cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_depth_images cl_khr_3d_image_writes cl_intel_exec_by_local_thread cl_khr_spir cl_khr_fp64 
  Platform Profile:				 FULL_PROFILE
  Platform Version:				 OpenCL 1.2 CUDA 7.5.23
  Platform Name:				 NVIDIA CUDA
  Platform Vendor:				 NVIDIA Corporation
  Platform Extensions:				 cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts 

  Platform Name:				 Intel Gen OCL Driver
Number of devices:				 1
  Device Type:					 CL_DEVICE_TYPE_GPU
  Device ID:					 32902
  Max compute units:				 20
  Max work items dimensions:			 3
    Max work items[0]:				 512
    Max work items[1]:				 512
    Max work items[2]:				 512
[... 184 lines cut ...]
  Global memory size:				 1073610752
  Constant buffer size:				 65536
  Max number of constant args:			 9
  Local memory type:				 Local
  Local memory size:				 49152
  Error correction support:			 0
  Unified memory for Host and Device:		 0
  Profiling timer resolution:			 1000
  Device endianess:				 Little
  Available:					 Yes
  Compiler available:				 Yes
  Execution capabilities:				 
    Execute OpenCL kernels:			 Yes
    Execute native function:			 No
  Queue properties:				 
    Out-of-Order:				 Yes
    Profiling :					 Yes
  Platform ID:					 0xd160e0
  Name:						 GeForce GT 730M
  Vendor:					 NVIDIA Corporation
  Device OpenCL C version:			 OpenCL C 1.2 
  Driver version:				 352.63
  Profile:					 FULL_PROFILE
  Version:					 OpenCL 1.2 CUDA
  Extensions:					 cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts  cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64

closed 2016-05-13 by wiedemeyer