code-iai/iai_kinect2 #283: only rgb stream possible?

flobotics, 2016-05-23:
hi,
is it possible to only use the rgb stream from the kinect2 ? when i only use the topic /kinect2/hd/image the kinect_bridge is also producing depth, which uses much cpu?

wiedemeyer, 2016-05-24:
Hi,
I just added some code to the devel branch to support this. Now it only starts the streams that are subscribed. Please check out the devel branch and test if everything is working as intended.

flobotics, 2016-05-24:
i did a "git checkout devel" and then run catkin_make. After running "roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true"  and then "rosrun rviz rviz" and select "camera" with topic "kinect2/hd/image" the kinect2_bridge still produces depth but the cpu usage is only around 75% (before ~95%)

[Info] [OpenCLDepthPacketProcessor] avg. time: 3.60028ms -> ~277.756Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 21.4186ms -> ~46.6883Hz
[ INFO] [1464091831.968545279]: [Kinect2Bridge::main] depth processing: ~0.12918ms (~7741.11Hz) publishing rate: ~29.9642Hz
[ INFO] [1464091831.968627688]: [Kinect2Bridge::main] color processing: ~23.441ms (~42.6602Hz) publishing rate: ~29.9642Hz

do i need any options to run kinect2_bridge ?

wiedemeyer, 2016-05-24:
Make sure that it got recompiled. Delete devel and build and run catkin_make again.
If I run kinect2_bridge and `rostopic hz /kinect2/hd/image_color` the output is:
```
[ INFO] [Kinect2Bridge::main] waiting for clients to connect
[ INFO] [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[ INFO] [Kinect2Bridge::main] color processing: ~7.64446ms (~130.814Hz) publishing rate: ~26.2493Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 4.31666ms -> ~231.661Hz
[ INFO] [Kinect2Bridge::main] color processing: ~7.62928ms (~131.074Hz) publishing rate: ~30.2391Hz
[Info] [VaapiRgbPacketProcessor] avg. time: 4.47766ms -> ~223.331Hz
```

flobotics, 2016-05-24:
the output of rostopic hz /kinect2/hd/image_color is :  
rostopic hz /kinect2/hd/image_color
subscribed to [/kinect2/hd/image_color]
average rate: 30.590
	min: 0.027s max: 0.040s std dev: 0.00521s window: 7
average rate: 29.987
	min: 0.022s max: 0.045s std dev: 0.00703s window: 37
average rate: 30.034
	min: 0.022s max: 0.047s std dev: 0.00710s window: 67

but from kinect2_bridge is still:
[ INFO] [1464093055.548026073]: [Kinect2Bridge::main] depth processing: ~0.125411ms (~7973.79Hz) publishing rate: ~29.6221Hz
[ INFO] [1464093055.548104568]: [Kinect2Bridge::main] color processing: ~18.5718ms (~53.8452Hz) publishing rate: ~26.2937Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.4472ms -> ~48.9065Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 8.7713ms -> ~114.008Hz

wiedemeyer, 2016-05-24:
Did you do a `git pull` after changing to devel?

flobotics, 2016-05-24:
no

closed 2016-05-24 by flobotics