code-iai/iai_kinect2 #285: "tf error"

kerolos, 2016-05-27:
i got this error when i run "publish_tf:=true" do you know how can fixed this error?
thanks in advance

$ roslaunch kinect2_bridge kinect2_bridge.launch reg_method:=opencl depth_methode:=opencl publish_tf:=true

 INFO] [1464350754.194748619]: [DepthRegistrationOpenCL::init]   0: GeForce GT 640M LE (GPU)[NVIDIA Corporation]
[ INFO] [1464350754.194829479]: [DepthRegistrationOpenCL::init] selected device: GeForce GT 640M LE (GPU)[NVIDIA Corporation]
[ INFO] [1464350754.272714031]: [Kinect2Bridge::callbackStatus] client connected. starting device...
[Info] [Freenect2DeviceImpl] starting...
/opt/ros/indigo/lib/nodelet/nodelet: symbol lookup error: /home/kerolos/catkin_ws/devel/lib//libkinect2_bridge_nodelet.so: undefined symbol: _ZN2tf20TransformBroadcasterC1Ev
[FATAL] [1464350754.324316585]: Service call failed!
[FATAL] [1464350754.324324922]: Service call failed!
[FATAL] [1464350754.325311805]: Service call failed!
[FATAL] [1464350754.325463093]: Service call failed!
[kinect2-1] process has died [pid 24383, exit code 127, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2-1.log].
log file: /home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2-1*.log
[kinect2_bridge-2] process has died [pid 24387, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load kinect2_bridge/kinect2_bridge_nodelet kinect2 __name:=kinect2_bridge __log:=/home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2_bridge-2.log].
log file: /home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2_bridge-2*.log
[kinect2_points_xyzrgb_sd-3] process has died [pid 24388, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/sd/camera_info rgb/image_rect_color:=kinect2/sd/image_color_rect depth_registered/image_rect:=kinect2/sd/image_depth_rect depth_registered/points:=kinect2/sd/points __name:=kinect2_points_xyzrgb_sd __log:=/home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2_points_xyzrgb_sd-3.log].
log file: /home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2_points_xyzrgb_sd-3*.log
[kinect2_points_xyzrgb_qhd-4] process has died [pid 24389, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/qhd/camera_info rgb/image_rect_color:=kinect2/qhd/image_color_rect depth_registered/image_rect:=kinect2/qhd/image_depth_rect depth_registered/points:=kinect2/qhd/points __name:=kinect2_points_xyzrgb_qhd __log:=/home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2_points_xyzrgb_qhd-4.log].
log file: /home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2_points_xyzrgb_qhd-4*.log
[kinect2_points_xyzrgb_hd-5] process has died [pid 24398, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb kinect2 rgb/camera_info:=kinect2/hd/camera_info rgb/image_rect_color:=kinect2/hd/image_color_rect depth_registered/image_rect:=kinect2/hd/image_depth_rect depth_registered/points:=kinect2/hd/points __name:=kinect2_points_xyzrgb_hd __log:=/home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2_points_xyzrgb_hd-5.log].
log file: /home/kerolos/.ros/log/e20ab6ee-2402-11e6-8fe1-5453edb62e3d/kinect2_points_xyzrgb_hd-5*.log
[kinect2_bridge-2] restarting process
process[kinect2_bridge-2]: started with pid [24675]

kerolos, 2016-05-27:
$ catkin_make

Base path: /home/kerolos/catkin_ws
Source space: /home/kerolos/catkin_ws/src
Build space: /home/kerolos/catkin_ws/build
Devel space: /home/kerolos/catkin_ws/devel
Install space: /home/kerolos/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/kerolos/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/kerolos/catkin_ws/build"
####
[  3%] [  6%] Built target pcl_create
Built target pcl_matching
[ 13%] [ 24%] [ 24%] [ 27%] Built target pcl_filter
Built target kinect2_registration
Built target pcl_convex_hull
[ 31%] [ 34%] Built target pcl_downsampling
Built target pcl_for_loop
Built target cluster_extract
[ 37%] [ 44%] [ 44%] Built target pcl_model_estimation
Built target pcl_partitioning
Built target pcl_read
[ 48%] Built target pcl_planar_segmentation
[ 51%] Built target pcl_project1
[ 55%] Built target pcl_project2
[ 58%] Built target pcl_multi_planes
[ 62%] Built target pcl_project3
[ 65%] [ 68%] [ 72%] Built target pcl_visualize
Built target pcl_simple_segmentation
Built target pcl_write
[ 79%] [ 79%] [ 82%] Built target pcl_visualize2
Built target plc_mult_pub
Built target pcl_regiongrowing_convex
[ 86%] Built target pcl_wall_extract
Linking CXX executable /home/kerolos/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
[ 89%] Built target kinect2_calibration
[ 93%] Built target kinect2_bridge_nodelet
[ 96%] Built target kinect2_viewer
CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o: In function `Kinect2Bridge::publishStaticTF()':
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge15publishStaticTFEv[_ZN13Kinect2Bridge15publishStaticTFEv]+0x2b): undefined reference to `tf::TransformBroadcaster::TransformBroadcaster()'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge15publishStaticTFEv[_ZN13Kinect2Bridge15publishStaticTFEv]+0x813): undefined reference to `tf::TransformBroadcaster::sendTransform(tf::StampedTransform const&)'
kinect2_bridge.cpp:(.text._ZN13Kinect2Bridge15publishStaticTFEv[_ZN13Kinect2Bridge15publishStaticTFEv]+0x823): undefined reference to `tf::TransformBroadcaster::sendTransform(tf::StampedTransform const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/kerolos/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge] Error 1
make[1]: *** [iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

wiedemeyer, 2016-05-27:
I guess your setup is probably the cause. Make sure everything is sourced correctly and that you have the ROS packages installed.

closed 2016-06-14 by wiedemeyer