code-iai/iai_kinect2 #286: how to convert point cloud data into pcd file ?

nomi321, 2016-05-28:
i want to convert point cloud data into pcd file. how can i do this ?

wiedemeyer, 2016-05-30:
[This should do the job.](http://lmgtfy.com/?q=ros+point+cloud+to+pcd)

nomi321, 2016-05-30:
bro i tried this already. but it converts a point cloud to multiple pcd files. is this correct or m making any mistake ? m sorry if my question made u laugh ... does all pcd files represnt that cloud ?

wiedemeyer, 2016-05-30:
From the [documentation](http://wiki.ros.org/pcl_ros#pcd_to_pointcloud) of `pointcloud_to_pcd`:
"Subscribes to a ROS topic and saves point cloud messages to PCD files. Each message is saved to a separate file, names are composed of an optional prefix parameter, the ROS time of the message, and the .pcd extension."
PCD: http://pointclouds.org/documentation/tutorials/pcd_file_format.php

nomi321, 2016-05-30:
ok thanks bro ...

closed 2016-05-30 by nomi321