code-iai/iai_kinect2 #288: catkin_make failure

ccrescent, 2016-06-01:
Hi, I can install the libfreenect2 successfully, and ./bin/Protonect works.
but when I try to compile the iai_kinect2 ,($ catkin_make -DCMAKE_BUILD_TYPE="Release") it shows that:

/usr/include/CL/cl.hpp :1041:38 error : '_mm_mfence' was not declared in this scope 
inline void fence() { _mm_mfence();

I have used the new version of libfreenect2
https://github.com/OpenKinect/libfreenect2 ;

what should i do to solve it .. @xlz

ccrescent, 2016-06-02:
now it works ,i mean i can't compile the iai_kinect2 .(I use the same cmd catkin_make -DCMAKE_BUILD_TYPE="Release",and use the backup iai_kinect2 (downloaded several days before))
but  why the above question will come before? Because I really try many times yestoday according the tutorials .

wiedemeyer, 2016-06-02:
Seems like the compiler didn't enabled SSE by default. The solutions in the following issues should help:
https://github.com/OpenKinect/libfreenect2/issues/250
https://github.com/OpenKinect/libfreenect2/issues/355

ccrescent, 2016-06-02:
Thank you , I will try  if it comes again  TAT . @wiedemeyer

closed 2016-06-04 by ccrescent