code-iai/iai_kinect2 #289: kinect2_calibration can't save frame

ccrescent, 2016-06-02:
Hi, 
in Detailed steps:
at 3.Record images for the color camera: rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color

when running the node above ,i can see the image ,but I can't save frames to kinect_cal_data .and it seems that the instructions don't work . (just input the instructions? )

the node shows:
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1464850290115448126/compressed/set_parameters]

(i have only run  roscore and kinect2_bridge)

T.T help

wiedemeyer, 2016-06-02:
The image window has to be active, the board has to be visible and detected, then press space to save images.

ccrescent, 2016-06-02:
I have actived the image window ,and the board is visible and detected . But i still can't see anything in kinect_cal_data ... I run the node in folder @wiedemeyer

ccrescent, 2016-06-03:
considering it may be my computer 's problems,  I changed another one for test . But I still can't save any file .("q" works).

ccrescent, 2016-06-03:
sorry , I have made some stupid mistakes .

closed 2016-06-03 by ccrescent

wiedemeyer, 2016-06-03:
Glad to hear that it is working.

Allen-jl, 2017-02-28:
@crescent 
I just met the same problem as yours, i have already activated the color window and press the space, but i still cannot save any images and the color window could detect the borad...
so, could anyone tell me how can i fix it? THX

YoungRainy, 2017-03-16:
@ccrescent 
me too ,i also met the same problem,cannot save images but "q"  works...
errors like:
[ERROR] Tried to advertise a service that is already advertised in this node [/kinect2_calib_1489672995873018352/compressed/set_parameters]

did anyone know how to fix this problem? Thanks.

YoungRainy, 2017-03-16:
@Allen-jl 
i got the reason,i used the wrong pattern ,as the command is :rosrun kinect2_calibration kinect2_calibration chess5x7x0.03 record color , you should use 5*7*0.03 pattern,then you will find colored overlay shown on the pattern, and images will be saved when i press "s".
stupid mistakes...

Coldplayplay, 2017-07-30:
hello, meet the same question, using the chess8x11x0.03 pattern, cannot save the image