code-iai/iai_kinect2 #290: Invoking "make -j4 -l4" failed

AyanGithub, 2016-06-03:
I am new to this Ubuntu and ROS stuff. In fact, I installed both of those today. What I'm puzzled on is this: `Invoking "make -j4 -l4" failed.`
This is the part of code.
`####
#### Running command: "make -j4 -l4" in "/home/ayan/catkin_ws/build"
####
Scanning dependencies of target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
Scanning dependencies of target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
Scanning dependencies of target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
Scanning dependencies of target kinect2_registration
[  1%] Building CXX object iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/kinect2_registration.cpp.o
[  1%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_b
[  1%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_s
[  1%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui_f
Scanning dependencies of target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
Scanning dependencies of target siftgpu_proj
[  2%] Building CXX object iai_kinect2/kinect2_registration/CMakeFiles/kinect2_registration.dir/src/depth_registration_cpu.cpp.o
[  3%] Creating directories for 'siftgpu_proj'
[  3%] Built target _rgbdslam_generate_messages_check_deps_rgbdslam_ros_ui
Scanning dependencies of target rgbdslam_generate_messages_cpp
[  4%] Generating C++ code from rgbdslam/rgbdslam_ros_ui_f.srv
[  6%] No download step for 'siftgpu_proj'
[  7%] No patch step for 'siftgpu_proj'
[  8%] No update step for 'siftgpu_proj'
[  9%] Generating C++ code from rgbdslam/rgbdslam_ros_ui_s.srv
[ 10%] No configure step for 'siftgpu_proj'
[ 12%] Performing build step for 'siftgpu_proj'
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[ 13%] Generating C++ code from rgbdslam/rgbdslam_ros_ui_b.srv
[ 14%] Generating C++ code from rgbdslam/rgbdslam_ros_ui.srv
[ 14%] Built target rgbdslam_generate_messages_cpp
Scanning dependencies of target rgbdslam_generate_messages_eus
[ 15%] Generating EusLisp code from rgbdslam/rgbdslam_ros_ui_f.srv
[ 17%] Generating EusLisp code from rgbdslam/rgbdslam_ros_ui_s.srv
[ 18%] Generating EusLisp code from rgbdslam/rgbdslam_ros_ui_b.srv
[ 19%] Generating EusLisp code from rgbdslam/rgbdslam_ros_ui.srv
[ 20%] Generating EusLisp manifest code for rgbdslam
Scanning dependencies of target rgbdslam_generate_messages_py
[ 21%] Generating Python code from SRV rgbdslam/rgbdslam_ros_ui_f
[ 23%] Generating Python code from SRV rgbdslam/rgbdslam_ros_ui_s
[ 24%] Generating Python code from SRV rgbdslam/rgbdslam_ros_ui_b
[ 25%] Generating Python code from SRV rgbdslam/rgbdslam_ros_ui
[ 26%] Generating Python srv __init__.py for rgbdslam
[ 26%] Built target rgbdslam_generate_messages_py
Scanning dependencies of target rgbdslam_generate_messages_lisp
[ 28%] Generating Lisp code from rgbdslam/rgbdslam_ros_ui_f.srv
[ 29%] Linking CXX shared library /home/ayan/catkin_ws/devel/lib/libkinect2_registration.so
[ 30%] Generating Lisp code from rgbdslam/rgbdslam_ros_ui_s.srv
[ 31%] Generating Lisp code from rgbdslam/rgbdslam_ros_ui_b.srv
[ 32%] Generating Lisp code from rgbdslam/rgbdslam_ros_ui.srv
[ 32%] Built target kinect2_registration
[ 32%] Built target rgbdslam_generate_messages_lisp
Scanning dependencies of target rgbdslam_generate_messages_nodejs
Scanning dependencies of target kinect2_bridge
[ 34%] Generating Javascript code from rgbdslam/rgbdslam_ros_ui_f.srv
[ 35%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge.dir/src/kinect2_bridge.cpp.o
[ 36%] Generating Javascript code from rgbdslam/rgbdslam_ros_ui_s.srv
[ 37%] Generating Javascript code from rgbdslam/rgbdslam_ros_ui_b.srv
[ 39%] Generating Javascript code from rgbdslam/rgbdslam_ros_ui.srv
[ 39%] Built target rgbdslam_generate_messages_eus
Scanning dependencies of target kinect2_bridge_nodelet
[ 39%] Built target rgbdslam_generate_messages_nodejs
Scanning dependencies of target kinect2_calibration
[ 40%] Building CXX object iai_kinect2/kinect2_bridge/CMakeFiles/kinect2_bridge_nodelet.dir/src/kinect2_bridge.cpp.o
[ 41%] Building CXX object iai_kinect2/kinect2_calibration/CMakeFiles/kinect2_calibration.dir/src/kinect2_calibration.cpp.o
/usr/bin/ld: cannot find -lIL
collect2: error: ld returned 1 exit status
makefile:170: recipe for target 'siftgpu' failed
make[3]: *** [siftgpu] Error 1
felixendres-rgbdslam_v2-bb400d7/external/CMakeFiles/siftgpu_proj.dir/build.make:110: recipe for target 'felixendres-rgbdslam_v2-bb400d7/external/siftgpu_prefix/src/siftgpu_proj-stamp/siftgpu_proj-build' failed
make[2]: *** [felixendres-rgbdslam_v2-bb400d7/external/siftgpu_prefix/src/siftgpu_proj-stamp/siftgpu_proj-build] Error 2
CMakeFiles/Makefile2:3953: recipe for target 'felixendres-rgbdslam_v2-bb400d7/external/CMakeFiles/siftgpu_proj.dir/all' failed
make[1]: *** [felixendres-rgbdslam_v2-bb400d7/external/CMakeFiles/siftgpu_proj.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 42%] Linking CXX executable /home/ayan/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
[ 42%] Built target kinect2_bridge
[ 43%] Linking CXX shared library /home/ayan/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so
[ 45%] Linking CXX executable /home/ayan/catkin_ws/devel/lib/kinect2_calibration/kinect2_calibration
[ 45%] Built target kinect2_bridge_nodelet
[ 45%] Built target kinect2_calibration
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
**Invoking "make -j4 -l4" failed**
`

AyanGithub, 2016-06-03:
I need an answer ASAP please!

wiedemeyer, 2016-06-05:
```
/usr/bin/ld: cannot find -lIL
collect2: error: ld returned 1 exit status
makefile:170: recipe for target 'siftgpu' failed
make[3]: *** [siftgpu] Error 1
```
This is an error related to siftgpu and not to iai_kinect2.

closed 2016-06-14 by wiedemeyer