code-iai/iai_kinect2 #292: Installation question

navakatou, 2016-06-07:
In the step number 4 I have done all this 
cd ~/catkin_ws/src/
git clone https://github.com/code-iai/iai_kinect2.git
cd iai_kinect2
rosdep install -r --from-paths 
Usage: rosdep [options] <command> <args>
Commands:
rosdep check <stacks-and-packages>...
  check if the dependencies of package(s) have been met.
rosdep install <stacks-and-packages>...
  generate a bash script and then execute it.
rosdep db
  generate the dependency database and print it to the console.
rosdep init
  initialize rosdep sources in /etc/ros/rosdep.  May require sudo.
rosdep keys <stacks-and-packages>...
  list the rosdep keys that the packages depend on.
rosdep resolve <rosdeps>
  resolve <rosdeps> to system dependencies
rosdep update
  update the local rosdep database based on the rosdep sources.
rosdep what-needs <rosdeps>...
  print a list of packages that declare a rosdep on (at least
  one of) <rosdeps>
rosdep where-defined <rosdeps>...
  print a list of yaml files that declare a rosdep on (at least
  one of) <rosdeps>
rosdep fix-permissions
  Recursively change the permissions of the user's ros home directory.
  May require sudo.  Can be useful to fix permissions after calling
  "rosdep update" with sudo accidentally.
rosdep: error: no packages or stacks specified

So I don't know why I'm getting this. I have installed PCL, libfreenect2,Eigen,  OpenCV 2.4.9 and I have checked the Protonect and it worked.

wiedemeyer, 2016-06-08:
Did you executed `sudo rosdep init` and `rosdep update` after you installed ROS? Also if you install `ros-indigo-desktop-full` all needed dependencies should be installed anyway.

closed 2016-07-07 by wiedemeyer

wendaoyuchen, 2017-04-07:
I know!! I solved this problem just now!
the command should be "rosdep install -r --from-paths ." rather than "rosdep install -r --from-paths"
the difference between them is just a point !

denglinan, 2019-03-09:
> I know!! I solved this problem just now!
> the command should be "rosdep install -r --from-paths ." rather than "rosdep install -r --from-paths"
> the difference between them is just a point !

haha, thank you, I make the same mistake as you.