code-iai/iai_kinect2 #295: kinect2_link Location on Device?

ViktorZielke, 2016-06-15:
Hi, 

I'm currently trying to add the Kinect into a tf-tree. The Kinect is mounted on a robotic arm. In relation to that I couldn't find any information regarding which point on the real device references the kinect2_link location. Is it the mounting point for the screw at the bottom/underside of the Kinect?

Greetings
Viktor

wiedemeyer, 2016-06-15:
It is the center of the rgb sensor, so somewhere inside the device. We did a arm to camera calibration to get the correct values for the URDF of our robots.

JKBehrens, 2016-06-17:
Hi,

have you used [1] or can you recommend a calibration tool?

Best regards,
Jan

[1] http://wiki.ros.org/industrial_extrinsic_cal

wiedemeyer, 2016-06-17:
We used Halcon for the calibration, and for the PR2 we used the pr2_calibration. I used too few calibration tools, to recommend one.

romarcg, 2016-06-28:
Hello,

I am having issues with the published tf from kinect2_bridge launch.

I have done the calibration of the kinect2 as suggested in the documentation. However, when I start the kinect_bridge (enabling the tf publishing and using the calibration data), the transformation between the camera base link and the rgb_optical_frame does not seem right:

rosrun tf tf_echo /kinect2_link /kinect2_rgb_optical_frame

- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]

I have verified the cloud registration after the calibration and it seems better than the default data, I am only having issues with the tf. Moreover, the transformation between the rgb and ir optical frames seems correct:

rosrun tf tf_echo /kinect2_rgb_optical_frame /kinect2_ir_optical_frame

- Translation: [-0.052, 0.000, 0.003]
- Rotation: in Quaternion [-0.002, 0.001, -0.001, 1.000]
            in RPY (radian) [-0.004, 0.002, -0.001]
            in RPY (degree) [-0.246, 0.140, -0.069]

Do you have any clues where the issue is coming from?

Thanks in advance

Best regards,

wiedemeyer, 2016-06-29:
This looks totally fine. There is no point where /kinect2_link should refer to, so it is the same as /kinect2_rgb_optical_frame by default.

romarcg, 2016-06-30:
Hello,

Thank you for your answer. I was using a calibration node for kinect1-to-robot setup and I did not realise that the tf for the kinect1, specially the base_link orientation, is not the same as the kinect2. What I did is to create a dummy tf node to change the kinect2_link orientation to the desired one. The setup is working properly now.

Thank you for the tools you have developed.

Nice work.

closed 2016-08-09 by wiedemeyer