code-iai/iai_kinect2 #296: Install error about "rosdep install -r --from-paths ."

QinZiwen, 2016-06-17:
I had already run "sudo rosdep init" and "rosdep update" as nice. But errors hit me when I run "rosdep install -r --from-paths ." 
===============================================
turtlebot-pc@turtlebot-pc-Veriton-B832:~/catkin_ws/src/iai_kinect2$ rosdep install -r --from-paths .
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
kinect2_viewer: Cannot locate rosdep definition for [kinect2_bridge]
kinect2_bridge: Cannot locate rosdep definition for [kinect2_registration]
kinect2_calibration: Cannot locate rosdep definition for [kinect2_bridge]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
===============================================
what shuld i do?

wiedemeyer, 2016-06-17:
Maybe reading the [README](https://github.com/code-iai/iai_kinect2#install):
*Note: `rosdep` will output errors on not being able to locate `[kinect2_bridge]` and `[depth_registration]`.
   That is fine because they are all part of the iai_kinect2 package and `rosdep` does not know these packages.*

closed 2016-06-17 by wiedemeyer

QinZiwen, 2016-06-22:
What should I do， my "rosdep" can know iai_kinect2 package

xiaobai333333, 2024-05-02:
> What should I do， my "rosdep" can know iai_kinect2 package
解决了么，怎么解决的？