code-iai/iai_kinect2 #297: run "rosdep install -r --from-paths ." error

QinZiwen, 2016-06-19:
When I was run "rosdep install -r --from-paths .", some errors show me as following:
=================================
turtlebot@turtlebot-HP-ZBook-14-G2:~/catkin_ws/src/iai_kinect2$ rosdep install -r --from-paths .
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
kinect2_viewer: Cannot locate rosdep definition for [kinect2_bridge]
kinect2_calibration: Cannot locate rosdep definition for [kinect2_bridge]
kinect2_bridge: Cannot locate rosdep definition for [kinect2_registration]
Continuing to install resolvable dependencies...
All required rosdeps installed successfully
=================================
Some assistes show as following  writing by author:
rosdep will output errors on not being able to locate [kinect2_bridge] and [depth_registration]. That is fine because they are all part of the iai_kinect2 package and rosdep does not know these packages.

what should I do, my rosdep can just know these packages.
Thank you very much!

wiedemeyer, 2016-06-20:
You don't have to open one issue twice. #296
This is ok, you can ignore it.

QinZiwen, 2016-06-21:
When I run "cmake ", some errors show me again:
===========================
turtlebot@turtlebot-HP-ZBook-14-G2:~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE="Release"
Base path: /home/turtlebot/catkin_ws
Source space: /home/turtlebot/catkin_ws/src
Build space: /home/turtlebot/catkin_ws/build
Devel space: /home/turtlebot/catkin_ws/devel
Install space: /home/turtlebot/catkin_ws/install
####
#### Running command: "cmake /home/turtlebot/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/turtlebot/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/turtlebot/catkin_ws/install -G Unix Makefiles" in "/home/turtlebot/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/turtlebot/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/turtlebot/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/turtlebot/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/turtlebot/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.11
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - kinect2_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - kinect2_viewer
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'kinect2_registration'
-- ==> add_subdirectory(iai_kinect2/kinect2_registration)
-- Could NOT find OpenCL (missing:  OpenCL_INCLUDE_DIR) 
-- CPU based depth registration disabled
-- OpenCL based depth registration disabled
CMake Error at iai_kinect2/kinect2_registration/CMakeLists.txt:77 (message):
  No registration method available!

-- Configuring incomplete, errors occurred!
See also "/home/turtlebot/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/turtlebot/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
======================================

wiedemeyer, 2016-06-22:
Install OpenCL, pull the latest changes, remove devel and build folders and run catkin_make again.

QinZiwen, 2016-06-23:
Yes, iai_kinect2 works nice now. Thank you very much @wiedemeyer . However I thking that OpenCL is not optional but integrant.

wiedemeyer, 2016-06-23:
No, it's not. You either have to have Eigen3 or OpenCL or both installed.

closed 2016-06-23 by wiedemeyer

QinZiwen, 2016-06-26:
o, I see, thank you again！