code-iai/iai_kinect2 #298: Kinect 2 Can't Get Any Visualization

juliastake, 2016-06-20:
Hello,
I have  2 problem: 
1)Whenever i start  to roslaunch kinect2_bridge kinect2_bridge.launch,  i got output from terminal,
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[TurboJpegRgbPacketProcessor] avg. time: 24.6787ms -> ~40.5207Hz

Usually most of the time (especially doing  more than one work ) all the output is
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
but 2-3 weeks ago i got easily waiting for clients to connect output as expected, what is wrong right now ?
(i am not starting any extra node other than kinect bridge btw)

2)I can't get any visualization on RVIZ, i can  see nodes names in the list for depthcloud or pointcloud but when i get subscribe //For pointcloud 2
from rviz For frame [/head_mount_kinect2_rgb_optical_frame]: Fixed Frame [map] does not exist
from terminal  MessageFilter [target=map ]: Dropped 100,00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
 
Intel Core i7-6500U
Intel HD Graphics 520
is it too low for handle the kinect2 ?

wiedemeyer, 2016-06-20:
1. You have an Intel GPU so you could try to use VAAPI, which is much faster then TurboJpeg.
  I don't understand the paragraph. More than one what? You don't get `waiting for clients...`?

2. Make sure to set the `Fixed Frame` in `Global Options` to `kinect2_link` and that you enable `publish_tf` in the launch file.

juliastake, 2016-06-20:
2)That's weird i could change to Global Options to kinect2_link but right now i cant see any other option other than " /map" 

1)What i mean is ,  i cant see any "waiting for clients" message right now but when I use the last time ( 2-3 weeks ago) i can get it. Right now Only mesages are;
[RgbPacketStreamParser::handleNewData] skipping rgb packet!
[TurboJpegRgbPacketProcessor] avg. time: ......................

or 
[TransferPool::submit] failed to submit transfer

Is there any "how-to" for using VAAPI with kinect2_bridge btw

wiedemeyer, 2016-06-20:
2. Enable `publish_tf`.
1. That means that some client is connected.

VAAPI: https://github.com/OpenKinect/libfreenect2#linux

onurtore, 2016-06-24:
Thank you Thiemo
My pointcloud image in Rviz looks like upside
[image]

Is it a problem of calibration because we didn't do any calibration.

And finally is there any quick command for using VAAPI with kinect2 bridge or should i look for documentation for libfreenect2

wiedemeyer, 2016-06-24:
For `sd` no calibration is needed, only `qhd` and `hd` needs calibration. The orientation is normal, because it is not defined. You have to publish a TF for `map` to `kinect2_link` to change it.
You will have to install VAAPI and recompile libfreenect2 and kinect2_bridge.

closed 2016-08-09 by wiedemeyer