code-iai/iai_kinect2 #299: Roslaunch Error Getting Pointcloud

dkorobov2, 2016-06-20:
Hi,

When I run `roslaunch kinect2_bridge kinect2_bridge.launch` I am able to correctly retrieve the rgb, depth, and ir from the kinect2. However, if I try to subscribe to a pointcloud (/kinect2/qhd/points) for example, the terminal where I ran the kinect2_bridge displays: 

`terminate called after throwing an instance of 'image_transport::TransportLoadException'
  what():  Unable to load plugin for transport 'raw', error string:
Could not find library corresponding to plugin image_transport/raw_sub. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[kinect2-1] process has died [pid 16872, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=kinect2 __log:=/home/ubuntu/.ros/log/d73c73ce-340b-11e6-9b3e-00044b5824ce/kinect2-1.log].
log file: /home/ubuntu/.ros/log/d73c73ce-340b-11e6-9b3e-00044b5824ce/kinect2-1*.log`

I have tried executing `sudo apt-get install ros-indigo-image-transport ros-indigo-image-transport-plugins`, but I am still experiencing this problem. Any solution to this?

wiedemeyer, 2016-06-20:
Maybe you also need those: `ros-indigo-compressed-depth-image-transport`, `ros-indigo-compressed-image-transport`.
I usually install `ros-indigo-desktop-full` and never had issues.

dkorobov2, 2016-06-20:
Looks like I have both of those already. I found a similar issue [here](http://answers.ros.org/question/50091/image-transport-issue-in-rviz-groovy-beta-1/). Maybe I should do a clean install of ROS?

wiedemeyer, 2016-06-23:
And, did it work?

dkorobov2, 2016-06-23:
Unfortunately not. I also tried compiling image-transport-plugins from source and recompiling iai_kinect2 but no luck. For the record, I am running it on the Jetson TX1 and only have `ros-indigo-desktop` installed.

When I try `rospack plugins --attrib=plugin kinect2_bridge` it does not output anything. I'm guessing the plugin is not linked to the package correctly? I am fairly new to ROS so I am not sure how to proceed.

wiedemeyer, 2016-06-24:
make sure that ROS is sources correctly, also check that ros-indigo-depth-image-proc and ros-indigo-image-proc are installed.
If it is still not working you could search the internet for similar issues.

dkorobov2, 2016-06-24:
I think everything is sourced and installed correctly, and the internet has not been very helpful. If anyone has any ideas about this issue please let me know! Thanks.

dkorobov2, 2016-07-01:
I'm still not sure what the problem was, but I was able to get it to work with a clean boot and reinstallation of all the libraries.

closed 2016-07-01 by dkorobov2

DanLiu0623, 2018-01-25:
@dkorobov2  Hi, how do you solve this problem, I check my rospackage use "rospack list",I found image_transport, compressed_image_transport, compressed_depth_image_transport are all installed.Please help me~ Thank you very much!