code-iai/iai_kinect2 #301: kinect2_bridge: Segmentation fault

bemery94, 2016-06-22:
Hi,

When I try to “$ rosrun kinect2_bridge kinect2_bridge”, is gives a segmentation fault. I have tested libfreenect2 by running `./Protonect`, `./Protonect cl`, `./Protonect gl`, `./Protonect cpu` and they all work correctly, producing correct images and throwing no errors.

I am using nvidia-352 drivers and installed cuda according to the instructions on the iai_kienct2 git page. I'm not sure if it's relevant, but in /etc/OpenCl/vendors I have nvidia.icd and also intel-beignet-x86_64-linux-gnu.icd.

I ran the kinect2_bridge node in gdb mode, here is the output:

`(gdb) run
Starting program: /home/user/iai_kinect_ws/devel/lib/kinect2_bridge/kinect2_bridge 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
[New Thread 0x7fffde94e700 (LWP 6622)]
[New Thread 0x7fffde14d700 (LWP 6623)]
[New Thread 0x7fffdd94c700 (LWP 6624)]
[New Thread 0x7fffdd14b700 (LWP 6629)]
[New Thread 0x7fffd7fff700 (LWP 6630)]
[New Thread 0x7fffd77fe700 (LWP 6631)]
[ INFO] [Kinect2Bridge::initialize] parameter:
        base_name: kinect2
           sensor: default
        fps_limit: -1
       calib_path: /home/user/iai_kinect_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: opencl
     depth_device: -1
       reg_method: opencl
       reg_device: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 2
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: false
     base_name_tf: kinect2
   worker_threads: 4
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: GeForce GTX 960 (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 960 (GPU)[NVIDIA Corporation]
[New Thread 0x7fffd6df3700 (LWP 6651)]
[New Thread 0x7fffc23ff700 (LWP 6652)]
[New Thread 0x7fffc1bfe700 (LWP 6653)]
[New Thread 0x7fffc13fd700 (LWP 6654)]
[New Thread 0x7fffc0bfc700 (LWP 6655)]
[New Thread 0x7fffb7fff700 (LWP 6656)]
[New Thread 0x7fffb77fe700 (LWP 6657)]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Warning] [OpenCLDepthPacketProcessorImpl] Your ocl-icd has deadlock bugs. Update to 2.2.4+ is recommended.
[New Thread 0x7fffb34fd700 (LWP 6658)]
[New Thread 0x7fffb2cfc700 (LWP 6659)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 507498242542
[Info] [Freenect2Impl] found 1 devices
[ INFO] [Kinect2Bridge::initDevice] Kinect2 devices found: 
[ INFO] [Kinect2Bridge::initDevice]   0: 507498242542 (selected)
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ INFO] [Kinect2Bridge::initDevice] starting kinect2
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[ INFO] [Kinect2Bridge::initDevice] device serial: 507498242542
[ INFO] [Kinect2Bridge::initDevice] device firmware: 4.0.3911.0
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff710d282 in cv::Mat::copyTo(cv::_OutputArray const&) const () from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4

(gdb) bt
#0  0x00007ffff710d282 in cv::Mat::copyTo(cv::_OutputArray const&) const () from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4
#1  0x000000000044abaf in cv::Mat::clone (this=0x7fffffffc828) at /opt/ros/indigo/include/opencv-3.1.0-dev/opencv2/core/mat.inl.hpp:643
#2  0x00000000004547c5 in Kinect2Bridge::initDevice (this=0x7fffffffc670, sensor="507498242542")
    at /home/user/iai_kinect_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:603
#3  0x000000000044f7eb in Kinect2Bridge::initialize (this=0x7fffffffc670)
    at /home/user/iai_kinect_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:305
#4  0x000000000044d3b1 in Kinect2Bridge::start (this=0x7fffffffc670) at /home/user/iai_kinect_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:152
#5  0x0000000000448916 in main (argc=1, argv=0x7fffffffd5f8) at /home/user/iai_kinect_ws/src/iai_kinect2/kinect2_bridge/src/kinect2_bridge.cpp:1603
(gdb) 
`

Let me know if you need any more information. Thanks for your help.

wiedemeyer, 2016-06-22:
[At the point where it crashed](https://github.com/code-iai/iai_kinect2/blob/master/kinect2_bridge/src/kinect2_bridge.cpp#L603), it just copies a matrix that is created some lines before. This shouldn't crash at all, I guess something with your system is wrong.

bemery94, 2016-06-22:
Thanks for the quick reply! I worked out the issue, it's my fault. It appears that the computer that I am using has OpenCv 2.4.8 installed but it also had the OpenCv3 ROS package installed. After removing the ROS package it all works fine

closed 2016-06-22 by bemery94