code-iai/iai_kinect2 #303: Rectified image withe black borders

miguelriemoliveira, 2016-07-07:
Hello,

I have run a kinet2 calibration as suggested in the readme.

I get some rectified images which have a black border around them. For example, see bellow:

[image]
[image]

I wanted to ask if this also occurs to your rectified images. I think I understand why these black borders appear ... its because of the distortion rectification ... but I seldom have seen this kind of cushion like distortion ... the kind of distortion I am used to seeing is a barrel type distortion ( [see here](https://en.wikipedia.org/wiki/Distortion_(optics)))

Anyway, do you also have these black regions?

Thanks,

Miguel

wiedemeyer, 2016-07-08:
Yes, that is normal. And if you undistort an image with a barrel distortion, it will result in black borders, that get bigger to the center, like this one.

miguelriemoliveira, 2016-07-08:
Ok, thanks for the feedback, I will close the issue.

One more question though: @wiedemeyer do you have any idea why this did not occur with the kinect 1 or the asus xtion (at least I don't recall it)? Different combinations of RGB and IR cameras? I know the Kinect 2 has a wider field of view ...

closed 2016-07-08 by miguelriemoliveira